kismetwireless/kismet #302: Average Location and Peak Location Same

Geofferey, 2020-09-16:
When logging devices the average and peak locations are never updated in web UI. One would expect the peak locations and avg locations to adjust over time based on signal strength. For example if a networking comes into range down the street it will show up in spot where it was first detected instead of adjusting the location as signal gets stronger. This makes precision mapping of networks impossible when it seems it could be.

kismetwireless, 2020-09-16:
I'll take a look, but the average location would never be appropriate for
any sort of precision mapping.  Arguably precision mapping is impossible,
period, with a single sample - you'd need multiple *simultaneous* location
samples of *the same packet*, because different packets, at different
rates, have wildly different effective signal levels.

If you want to do more precise mapping, you'll want to pull the per-packet
signal levels out of the kismetdb log and manipulate that data.

That said, I'll still look and confirm that the peak signal level is
pulling from the packet w/ the peak signal.

dragorn, 2020-09-16:
As near as I can see the peak is correctly taken as the 'best' signal with
a literal single GPS value, from the packet with the best signal;
devicetracker_component.cc in the append_signal method.

If you only have a single packet which is counted towards the average, or
the only packets are at the exact same location (due to gps not refreshing
yet, etc) then they may come out equal.

The average location in the device record is a running average, which is
about the worst option for total accuracy but the only option available for
the runtime ui; for mapping offline, processing the packets is the way to
go.  There may be an API in the future in the rest interface to essentially
post-process in runtime by querying the database log, instead of trying to
hold the packets in ram, but there is currently no ETA on that.

If you've got data where you can show it's taking the wrong peak tho, let
me know.

-m

Geofferey, 2020-09-17:
Looking deeper, I think the problem likely due to Kismet not using full GPS coordinates from my app that's providing over TCP. Here is some screen shots to show what I mean. 

[image]

dragorn, 2020-09-17:
Not sure; would need to see the original data coming from the gps to be
able to replay it and see how it's interpreted.

Geofferey, 2020-09-17:
Try using ShareGps (TCP not GPSD)  from Google Play with Create NMEA checked in settings. I also tried Bluetooth GPS Output  Android app as well and see the same exact issue with LAT/LONG where last couple of degrees are missing. Does your Kismet show full GPS coordinates with whatever adapter you're using?

kismetwireless, 2020-09-19:
I think this is resolved by 8f1fc11c165bce075ebdf6d3dab2482c1ae8ec39
and 1a9798e5ddb12988a6b2c5450af8acdfb3d50996, and possibly has a little
help from 1e47bf24d50a249232f39b3fead524ee2bda9186.

-m

Geofferey, 2020-09-21:
@kismetwireless would this be in the latest nightlies for buster or would I need to compile it ?

closed 2020-09-26 by Geofferey