kismetwireless/kismet #303: ERROR: All configured channels are in error state!

rameshputrevu, 2020-09-18:
This is for Raspberry Pi 4. I tried to install from repo and I ran into issues, so I compiled from source code and installed.  But now when I start Kismet I see the following errors.

ERROR: All configured channels are in error state!
ERROR: Data source 'mon0 / mon0' ('mon0') encountered an error: Fatal 
       error from remote source
ALERT: SOURCEERROR Source mon0 has encountered an error. Kismet will 
       attempt to re-open the source in 5 seconds.  (1 failures)

iwdev output

eth0      no wireless extensions.
vethc9597d3  no wireless extensions.
mon0      IEEE 802.11  Mode:Monitor  Frequency:5.26 GHz  Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on          
br-bd98ec48a748  no wireless extensions.
vethd2154e6  no wireless extensions.
wlan0     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on 
vethb9e8aef  no wireless extensions.
docker0   no wireless extensions.
lo        no wireless extensions.
vethc8881f3  no wireless extensions.

Appreciate your help!

kismetwireless, 2020-09-18:
You must have the nexmon patches - either running kali, or manually
patching the kernel with the nexmon stuff.  You don't mention what you're
running or if you've applied them, but I'd start there.

rameshputrevu, 2020-09-18:
Thank you for the quick help!

My Pi4 is on Kernel 5.4.51-v7l+ and I applied the nexmon patch. Only after the nextmon patch, I was able to put the pi in monitor mode. see below mon0 mode: Monitor.. 

mon0 IEEE 802.11 Mode:Monitor Frequency:5.26 GHz Tx-Power=31 dBm.

Initially I tried to install kismet from the kismet repo and I ran into issues. That's when I got the sourcecode and compiled on raspberry pi 4b and installed using 'sudo make suidinstall'.

Now when I run Kismet from raspberry pi, I get the errors..

dragorn, 2020-09-18:
Unfortunately I don't know then; i can't replicate on a pi4 here with the
kali variants of the patches at least.

rameshputrevu, 2020-09-18:
The issue for me could be with setting up kismet itself.. as I tried to install Kismet and ran into various issues..that's when I gave up and built the kismet from source on my pi4.. may be in the process I screwed up kismet install.. 

can any one tell me how to clean up/remove/uninstall the half baked kismet install from the repo and as the install I built using sources.. I will do all over again and see if it helps..