kismetwireless/kismet #306: apt cannot upgrade from ubuntu repository

sliddjur, 2020-09-21:
OS: Ubuntu 20.04
APT Repository: `deb https://www.kismetwireless.net/repos/apt/release/focal focal main`

It seems like Ubuntu apt doesn't like version naming.

Then I tried to do a workaround with uninstalling kismet, and reinstalling. But I see now that R2 is no longer available. And `dpkg` still complains version number doesnt start with a digit.

```
olof@olof-ubuntu ~> apt list --upgradable 
Listing... Done
insync/unknown 3.2.7.40868-focal amd64 [upgradable from: 3.2.6.40863-focal]
kismet-adsb-icao-data/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet-capture-linux-bluetooth/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet-capture-linux-wifi/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet-capture-nrf-51822/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet-capture-nrf-mousejack/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet-capture-nxp-kw41z/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet-capture-ti-cc-2531/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet-capture-ti-cc-2540/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet-capture-ubertooth-one/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet-core/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet-logtools/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
kismet/focal kismet-2020-09-R3 amd64 [upgradable from: 2020-09-R2]
libc++1/focal 1:10.0-50~exp1 amd64 [upgradable from: 1:9.0-49~exp1]
python3-kismetcapturefreaklabszigbee/focal,focal kismet-2020-09-R3 all [upgradable from: 2020-09-R2]
python3-kismetcapturertl433/focal,focal kismet-2020-09-R3 all [upgradable from: 2020-09-R2]
python3-kismetcapturertladsb/focal,focal kismet-2020-09-R3 all [upgradable from: 2020-09-R2]
python3-kismetcapturertlamr/focal,focal kismet-2020-09-R3 all [upgradable from: 2020-09-R2]
```

And when I try to upgrade
```
olof@olof-ubuntu ~> sudo apt upgrade -y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libc++1
The following packages will be upgraded:
  insync kismet kismet-adsb-icao-data kismet-capture-linux-bluetooth kismet-capture-linux-wifi kismet-capture-nrf-51822 kismet-capture-nrf-mousejack kismet-capture-nxp-kw41z
  kismet-capture-ti-cc-2531 kismet-capture-ti-cc-2540 kismet-capture-ubertooth-one kismet-core kismet-logtools python3-kismetcapturefreaklabszigbee python3-kismetcapturertl433
  python3-kismetcapturertladsb python3-kismetcapturertlamr
17 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Need to get 170 MB of archives.
After this operation, 1 739 kB of additional disk space will be used.
Get:1 http://apt.insync.io/ubuntu focal/non-free amd64 insync amd64 3.2.7.40868-focal [145 MB]
Get:2 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-core amd64 kismet-2020-09-R3 [4 094 kB]
Get:3 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-adsb-icao-data amd64 kismet-2020-09-R3 [6 441 kB]
Get:4 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-capture-linux-wifi amd64 kismet-2020-09-R3 [142 kB]
Get:5 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-capture-linux-bluetooth amd64 kismet-2020-09-R3 [99,9 kB]
Get:6 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-capture-nrf-mousejack amd64 kismet-2020-09-R3 [90,8 kB]
Get:7 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 python3-kismetcapturertl433 all kismet-2020-09-R3 [52,5 kB]
Get:8 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 python3-kismetcapturertladsb all kismet-2020-09-R3 [65,5 kB]
Get:9 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 python3-kismetcapturertlamr all kismet-2020-09-R3 [60,5 kB]
Get:10 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 python3-kismetcapturefreaklabszigbee all kismet-2020-09-R3 [53,0 kB]
Get:11 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-capture-ti-cc-2531 amd64 kismet-2020-09-R3 [91,1 kB]
Get:12 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-capture-ubertooth-one amd64 kismet-2020-09-R3 [90,6 kB]
Get:13 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-capture-nrf-51822 amd64 kismet-2020-09-R3 [86,5 kB]
Get:14 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-capture-nxp-kw41z amd64 kismet-2020-09-R3 [90,2 kB]
Get:15 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-logtools amd64 kismet-2020-09-R3 [13,5 MB]
Get:16 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet amd64 kismet-2020-09-R3 [1 138 B]                                                                    
Get:17 https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 kismet-capture-ti-cc-2540 amd64 kismet-2020-09-R3 [91,1 kB]                                                 
Fetched 170 MB in 17s (9 799 kB/s)                                                                                                                                                         
Preconfiguring packages ...
(Reading database ... 290031 files and directories currently installed.)
Preparing to unpack .../00-insync_3.2.7.40868-focal_amd64.deb ...
Unpacking insync (3.2.7.40868-focal) over (3.2.6.40863-focal) ...
dpkg: error processing archive /tmp/apt-dpkg-install-JmzIq2/01-kismet-core_kismet-2020-09-R3_amd64.deb (--unpack):
 parsing file '/var/lib/dpkg/tmp.ci/control' near line 2 package 'kismet-core':
[... 80 lines cut ...]
 /tmp/apt-dpkg-install-JmzIq2/11-kismet-capture-ubertooth-one_kismet-2020-09-R3_amd64.deb
 /tmp/apt-dpkg-install-JmzIq2/12-kismet-capture-nrf-51822_kismet-2020-09-R3_amd64.deb
 /tmp/apt-dpkg-install-JmzIq2/13-kismet-capture-nxp-kw41z_kismet-2020-09-R3_amd64.deb
 /tmp/apt-dpkg-install-JmzIq2/14-kismet-logtools_kismet-2020-09-R3_amd64.deb
 /tmp/apt-dpkg-install-JmzIq2/15-kismet_kismet-2020-09-R3_amd64.deb
 /tmp/apt-dpkg-install-JmzIq2/16-kismet-capture-ti-cc-2540_kismet-2020-09-R3_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Tried looking for older version of kismet. But now the R2 version is not available in repository.
```
olof@olof-ubuntu ~> apt-cache madison kismet
    kismet | kismet-2020-09-R3 | https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 Packages
    kismet | 2016.07.R1-1.1~build2 | http://ubuntu.mirror.su.se/ubuntu focal/universe amd64 Packages
olof@olof-ubuntu ~> sudo apt install kismet=2020-09-R2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Version '2020-09-R2' for 'kismet' was not found
olof@olof-ubuntu ~> sudo apt install kismet=kismet-2020-09-R2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Version 'kismet-2020-09-R2' for 'kismet' was not found
```

kismetwireless, 2020-09-21:
I'll re-generate the packages, try again in 4-5 hours from now.

The new packages should (once they're built) show up as -R3a

-m

Tibladar, 2020-09-21:
Same issue here.

Also, the [changelog on the website](https://www.kismetwireless.net/release/kismet-2020-09-R3/) doesn't exist.
When looking through the [changelog for R2](https://www.kismetwireless.net/release/kismet-2020-09-R2/) there seems to be an issue...
[image]

dragorn, 2020-09-21:
Like I said.  I'm regenerating the packages.  Try back in several hours.

Tibladar, 2020-09-21:
Upgrading works now.

Thanks!

closed 2020-09-21 by kismetwireless