kismetwireless/kismet #308: kismet dont work properly

giotsap, 2020-09-24:
I am trying to work 

Kismet for educational reasons. I use in Ubuntu 20.04.1 LTS but when i When I start it, i I cannot see the wifi networks. Can anyone help me?

[image]

kismetwireless, 2020-09-24:
While you don't provide much information about what your actual problem is,
the ath10k card is known to perform very poorly, to the point that it is
essentially unusable, as documented here:
https://kismetwireless.net/docs/readme/datasources_wifi/#supported-hardware
under the "devices known to have issues"

giotsap, 2020-09-24:
My problem is that i cant see wi-fi networks nor their packets.

[image]

Is there a way to fix that problem using this computer with ath10k?