kismetwireless/kismet #311: To configure datasources (or perform actions like starting, stopping, or listing available interfaces) you need to be logged in as the Kismet administrator.

tdwong, 2020-09-29:
Not sure what is going on, but I am constantly getting this error message every time I tried to LOCK the scan frequency.

This could be an usage error, but who can tell me what is an "administrative account" and how do I create one?

A username/password was created when I first access Kismet via Web interface and all subsequent access were successfully signed in with that username/password pair, but the issue persists.

Here is my configuration - A pre-built Kali image running on Raspberry Pi 3B+ with everything factory default.

kali@kali:/etc/kismet$ lsb_release -a
No LSB modules are available.
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	2020.3
Codename:	kali-rolling
kali@kali:/etc/kismet$ kismet --version
Kismet 2020-09-R3

kismetwireless, 2020-09-29:
You're probably running 2020-09-R3 - Hard to tell from the partial error
paste, but I believe this is fixed in 2020-09-R4.

tdwong, 2020-09-29:
@kismetwireless  Thanks for the reply.  I followed the instructions from https://www.kismetwireless.net/docs/readme/packages/ and upgraded kismet to 2020-09-R4.  The problem no longer exists.

kali@kali:~$ kismet --version
Kismet 2020-09-R4

Feel free to close this issue.

closed 2020-09-29 by tdwong