kismetwireless/kismet #312: Can't get Kismet to display Wifi Networks

Ares500, 2020-10-16:
Hi,

I've been trying to make kismet work for the last week but can't for some reason. I'm using a tp-link 722n adapter for which I have installed the drivers to make monitor mode work. I have even tested this with the airodump-ng command and no issues there. Nevertheless, when I launch kismet using the kismet -c wlan0 command, kismet boots up perfectly, no error messages appear, bt when I open the UI on my browser there are no networks shown. I'm using kali linux in virtual box and am overall new to all of this.[
[image]
](url)

dragorn, 2020-10-16:
All the versions of that card after v1 use a chipset with no standard
drivers.

As far as I know the only drivers do not implement monitor mode, so
unfortunately there is no way to use it with kismet.

Ares500, 2020-10-16:
I have installed drivers that do implement and enable monitor mode and packet injection... This was even tested so the problem can't be with the card.
[image]
[image]

dragorn, 2020-10-16:
OK; I don't have access to any such cards and am not familiar with any
functional drivers, so you'll have to figure out how it goes into monitor
mode; if it responded to any of the normal methods, Kismet would have
already enabled it.  If you figure it out, patches welcome; the control
code is all found in the capture_linux_wifi directory and it already tries
netlink + vif, netlink + change mode non-vif, and legacy ioctl modes; you
may need to add specific hooks for whatever brokenness the driver
implemented.

Ares500, 2020-10-16:
I am very new to all of this could you care to explain please which commands I have to run to get this fixed?

kismetwireless, 2020-10-16:
Since I don't have one of those cards, I don't know what you'd need to do
to make it work - whatever drivers you have found are non-standard and
don't work with the standard methods of getting monitor mode working.

Ares500, 2020-11-02:
I installed this driver https://github.com/aircrack-ng/rtl8188eus. What do I have to do to make kismet work with it?

Geofferey, 2020-11-02:
Have you tried enabling monitor mode manually? I have that dongle as well and had to enable monitor mode manually on the device I am running it on. 

     sudo airmon-ng check-kill
     sudo ip link set <interface> down
     sudo iw dev <interface> set type monitor