kismetwireless/kismet #313: Bluetooth datasource not working

anthok, 2020-10-19:
Using a Sena UD100 the hci comes up and appears to start correctly with no errors. However no data is actually seen by kismet. 

I’ve verified the adapter is working by using the same adapter with bluelog and that utility is working fine. I’ve tried with Kismet 2019-R2 and 2020-R4, same results.

kismetwireless, 2020-10-19:
Try the latest nightly builds, other than that unfortunately I have no
suggestions unless you're able to find any particular steps that cause it
to fail; I have several here running on multiple platforms with no issues
so I can't recreate a failure to debug it.

anthok, 2020-10-19:
I’ll try the nightly, I’m on Ubuntu 20.04 if that helps

anthok, 2020-10-19:
No luck on nightly, what info would help?

anthok, 2020-10-23:
Looks like the issue is related to how Kismet tells the bluetooth controller to start discovery. For whatever reason my UD100 is not actually going into discovery mode when entering adding a data source, this can be verified by using `bluetoothctl` and checking the controller state.

The fix for me
With already Kismet running (and attempting capture on hci0), enter `bluetoothctl` and manually start discovery, then data  will start showing up in Kismet.

Tried bluez 5.50 and 5.53, same issue on both.

dragorn, 2020-10-23:
Ahhh, interesting!  That absolutely gives me something to go on, I'll try
to automate that; i've been trying to get various system to exhibit the
problem without luck, that sounds like something that hopefully can be
automated.

RFDeck, 2026-11-01:
"No errors, but no discovery until it's started by hand in bluetoothctl" matches how the source set up the controller
back then. It sent READ_INFO, SET_POWERED and START_DISCOVERY once and never tried again. A START_DISCOVERY answered
with status 0x0A (busy), as can happen while bluetoothd is still busy with the controller, was dropped without a
message. 5f64afcf (2026-10-21, "linuxbluetooth: keep retrying setup until the controller is discovering") redoes the
setup every 2 s until a DISCOVERING event arrives. 8345dea6 (2026-10-20) also fixes the device-found events being
read 6 bytes short and restarts after a controller reset. Both are in kismet-2026-10-R1.

That busy was the cause here is an assumption, since there are no status codes in the thread. Could you retest with
R1? If discovery still doesn't start, `btmon` output while the source opens would show the answer to START_DISCOVERY.