kismetwireless/kismet #314: Kismet uses both TX/RX packets associated with device MAC to describe rssi of wlan clients

Geofferey, 2020-10-22:
For awhile now since v2019 I have noticed a weird issue in Kismet where devices that show up seemingly report the RSSI of their parent AP and vice versa. In current signal RSSI this appears as a flip-flopping between clients rssi and it's associated AP. This also causes the client or AP to show the max signal of whichever device you were closest to while also causing them to report false minimum signal as well.

To replicate this issue start kismet near your AP or an associated client non-hopping on channel matching AP and note the current and max RSSI of device closest to you. Now take a look at other devices associated with the AP and you may notice many of the clients will show a max RSSI closely resembling the APs current RSSI & if you were closer to a client  with stronger RSSI than AP you'll see that APs max RSSI resembling the client you were closest too. 

After further investigation (to rule out driver issues) using wireshark and wlan.ta ==  (transmit address)/wlan.ra (receive) filters it appears that kismet uses RSSI from both source and destination packets containing a particular client/APs MAC instead of just using the RSSI from sources transmitted packets. 

I have published a video on YouTube that shows the issues and what I am think is causing it. 

https://youtu.be/jj5lt-TAWFM

After thinking on this a bit more I realize the behavior may be slightly intentional... At same time I am pretty sure it would be possible and more appropriate to have the UI only display RSSI for  packets that were actually transmitted from clients/APs without dropping anything minus bridged devices etc.

kismetwireless, 2020-10-22:
I'm not going to have a chance to watch a video for a while, can you
summarize where in the code you think it is?

-m

Geofferey, 2020-10-23:
@kismetwireless honestly I wish I could just tell you where in code this is happening but it's far too complicated for me. All I know for sure is the web UI is also using packets destined to a device to derive it's RSSI in the UI instead of just using the packets it transmitted.

Geofferey, 2020-10-24:
Is this is the code responsible for getting the last_signal? If so, couldn't there also be a check to see if the packet originated from device before setting? Just speculating, it really is over my head. 

    void  kis_tracked_signal_data::append_signal(const kis_layer1_packinfo& lay1, bool update_rrd, time_t rrd_ts) {
    if (lay1.signal_type == kis_l1_signal_type_dbm && (sig_type == 0 || sig_type == 1)) {
        if (sig_type == 0) {
            signal_type->set("dbm");
            sig_type = 1;
        }

        if (lay1.signal_dbm != 0) && {
            last_signal->set(lay1.signal_dbm);

            if (min_signal->get() == 0 || min_signal->get() > lay1.signal_dbm) {
                min_signal->set(lay1.signal_dbm);
            }

            if (max_signal->get() == 0 || max_signal->get() < lay1.signal_dbm) {
                max_signal->set(lay1.signal_dbm);
            }

            if (update_rrd)
                get_signal_min_rrd()->add_sample(lay1.signal_dbm, rrd_ts);
        }

dragorn, 2020-10-24:
Try the latest git, or tomorrows nightly git packages;
signal/location/enrcryption/etc are now only applied to the source device
of a packet (which means they only apply to the bssid when the packet is
from the bssid itself).  This means some inferred devices (a destination
device that has not yet transmitted itself) will not get signal/location
updates, but that's probably more accurate than classifying all the devices
mentioned in a packet at once.

Geofferey, 2020-10-24:
@dragorn You sir are a god! Can't tell you how happy I am to see this! Just tested it out and seems to be working exactly as expected.

closed 2020-10-25 by kismetwireless