kismetwireless/kismet #316: Feature Request - API key generation

asmodianx, 2020-10-26:
The only way I see to interact with the Kismet API is using basic HTTP authentication.  There is mention of using an API key or session token but I see no way to generate an API key or token for scripting or remote integration.  Granted using a calculated Authorization: Basic xxxxxxxxxxxxxxxxxxxxxxx isn't too terrible but im having a rough time with the graylog json query input feature and I figured I would ask to see if api keys were on the road map.

kismetwireless, 2020-10-26:
There hasn't even been a release with the new webserver code yet. You're
welcome to use git master but it is the bleeding edge development code.

Once it's feature complete there will be a release. There is currently no
eta.

asmodianx, 2020-10-27:
Thanks, so its far off a roadmap item but no ETA.  Feel free to file as a feature request or to close if this is already planned and tracked else where.

RFDeck, 2026-11-08:
This exists now, also in 2026-10-R1. A logged-in admin can create, list and revoke API keys:

```sh
# create a read-only key that expires in a day (duration in seconds, 0 = no expiry)
curl -u admin:<password> -X POST \
  -d 'json={"name":"script","role":"readonly","duration":86400}' \
  http://localhost:2501/auth/apikey/generate.cmd

curl -u admin:<password> http://localhost:2501/auth/apikey/list.json
curl -u admin:<password> -X POST -d 'json={"name":"script"}' http://localhost:2501/auth/apikey/revoke.cmd
```

A script then sends the key as `?KISMET=<key>` or as the `KISMET` cookie; master also takes
`Authorization: Bearer <key>`. Key creation can be turned off with `httpd_allow_auth_creation=false`.

One thing I noticed while trying it: the list shows `kismet.httpd.auth.expiration` as 0 for every key, even for keys
with a duration. The keys do expire on time; only the list is wrong. I'll send a small PR for that.

Can this be closed?