kismetwireless/kismet #317: APs/Clients Deriving "Channel" from Neighbors and Shared Hardware

Geofferey, 2020-10-27:
After changing how signal is derived (works very nicely btw) a new issue has surfaced. 

When channel hopping is enabled 2.4GHz-5Ghz APs seem to show main channels from shared hardware and neighboring APs. I have made some screenshots to highlight this issue. 

[image]
[image]

Geofferey, 2020-10-27:
Here's trying to give you something to go off of. I commented out these two blocks of code and the problem has disappeared with cams. I don't know implication of doing this and it probably breaks something else, but looks like part of problem is here.

        //Suspect #2

        /*if (dot11info->dest_mac != dot11info->source_mac &&
                dot11info->dest_mac != dot11info->bssid_mac &&
                dot11info->dest_mac != globalreg->empty_mac &&
                !(dot11info->dest_mac.bitwise_and(globalreg->multicast_mac)) ) {

            dest_dev =
                d11phy->devicetracker->update_common_device(commoninfo,
                        dot11info->dest_mac, d11phy, in_pack,
                        (update_flags | UCD_UPDATE_PACKETS | UCD_UPDATE_SEENBY),
                        "Wi-Fi Device");
        }*/

        //Suspect #3

        /*if (dot11info->other_mac != dot11info->source_mac &&
                dot11info->other_mac != dot11info->dest_mac &&
                dot11info->other_mac != dot11info->bssid_mac &&
                dot11info->other_mac != globalreg->empty_mac &&
                !(dot11info->other_mac.bitwise_and(globalreg->multicast_mac)) ) {

            other_dev =                                                                                          d11phy->devicetracker->update_common_device(commoninfo,                                                  dot11info->other_mac, d11phy, in_pack,
                        (update_flags | UCD_UPDATE_PACKETS | UCD_UPDATE_SEENBY),                                                             "Wi-Fi Device");
        }*/

Been running on hopping for about 20 min and still hasn't shown error with channel

[image]

[image]

dragorn, 2020-10-27:
That removes associating target devices, so yea - that's going to make
devices disappear and not be identified, because you removed the ability to
identify them.

Neither of those blocks sets the frequency.

It's entirely normal to see client devices hopping around on different
frequencies; you're going to need to go through your pcaps and see what
they're doing on the wifi layer and on the radiotap signal header to see
what frequencies each individual packet is on.

Geofferey, 2020-10-27:
Is it normal to see client devices that have absolutely zero hardware support for 5GHz to show that they have been on a 5GHz channel? I'd get what you're saying if they did support 5Ghz but these devices like cams and lights do not have to ability to communicate on that band. Id also understand if the cameras AP was operating on a 5Ghz as well but it doesn't and the cams still  end up as operating under 5Ghz channels. 

Let's say hypothetically that the camera did have some kind of hidden 5ghz capabilities. Even then the packet distribution shows over 90% on 2.462 (CH11) so why does UI end up showing channel 48 and getting stuck on that because it supposedly saw a small bit of traffic on that freq? In my mind the channel column / last seen channel should either match the last seen channel or be derived based on the packet distribution. Basically it would make sense (to me) if it showed same info as main frequency but in channel format. 

Seriously tho, after removing that block of code it stopped with clients like cams but not APs. I add back and cams end up showing control freq 48 and not 11 which they mostly operate on. I know it doesn't make any sense because I can see it doesn't have the UCD_UPDATE_FREQUENCIES in it so how it's effecting this I have zero idea. 

Anyways I didn't noticed any issues with this until the signal stuff was fixed. Sadly this is a game code likes to play lmfao .

kismetwireless, 2020-10-27:
it's normal if your card is lying and reporting them on 5ghz because it's
got sync problems with the channel set, that's for sure - which is why you
need to look at the packets and see what's being reported by the card in
the radiotap signal headers.

It's also normal if you have a 5ghz AP bridging broadcast packets from the
device onto 5ghz.  Which you also need to look at the pcap and see if it's
only on broadcast packets from the AP, but that code explicitly does NOT
set signal levels on the inferred destination device.

>

Geofferey, 2020-10-27:
I did take a pcap like you said, I dont know exactly what I should be looking for to help you but I did upload it 

https://file.io/X6JdCeLc2X8f

If this is an issue with my driver/card tho why would it just be showing up now? This did not happen before. Also it could be combated by doing what I said and having the last seen channel be based off the same "main freq" that's showing in shots. It wouldn't truly be last seen anymore but at least it would show correct the most seen freq (main freq) in channel column when sorting on first page. 

The data is contradicting itself just based on the screenshots I uploaded and that's what I am not liking. The devices in shots are clearly operating 90% in 2.4ghz band based on data even if there is a bit of false information caused by a wonky card/driver/hopping. So, why not just have last channel be derived from main frequency info then?

Geofferey, 2020-10-30:
I'm pretty sure I've found the block of code that is responsible for the issues I've been experiencing with the channel column. 

     if (dot11info->channel != "0" && dot11info->channel != "") {
                bssid_dev->set_channel(dot11info->channel);
            } else if (pack_l1info != NULL && (pack_l1info->freq_khz != bssid_dev->get_frequency() ||
                    bssid_dev->get_channel() == "")) {
                try {
                    bssid_dev->set_channel(khz_to_channel(pack_l1info->freq_khz));
                } catch (const std::runtime_error& e) {
                    ;
                }
            }

After commenting out all blocks containing `if (dot11info->channel != "0" && dot11info->channel != "")` The column is probably somewhere in the realm of 95-90% accuracy. This certainly breaks the channel field in device page > device info > Channel (?) > "The phy-specific channel of the device, if known...." but there is huge improvement of accuracy in nearly any other field that shows a channel, especially the column.

kismetwireless, 2020-10-31:
As you note, that removes channel setting, so, yes - it will stop showing
channels, but I don't consider that a viable change to the code since it
stops ALL channel handling, and removes all classification of channels on
non-2.4ghz devices.

You don't seem to want to believe it or look at the pcaps, but here's proof
of the exact thing I'm talking about re: wi-fi drivers reporting the wrong
frequency for received packets, and theres nothing Kismet can do about it
(attached); that's with the mediatek in-kernel drivers which are otherwise
excellent, other drivers are likely even less reliable.

Geofferey, 2020-11-02:
You obviously do not fully read these.... maybe you don't have enough time, idk. 

Instead of replying by email actually go and look what I have written here because it is constantly edited. I don't just go around making this stuff up to troll you, yet you seem to think I'm full of it. I am more than willing to look at PCAPs but it kinda helps if I know what to look for, which I don't. Also, I could fully believe it was a driver error if devices/APs weren't showing the EXACT same control channels as neighboring APs and shared hardware in the column/info page... There is a direct correlation between what I am seeing and the signal fixes 0a8e67f9. There is also a direct correlation in the crossing of displayed channels, not just some randomness or adjacent channel in the overlap. Anyways, for the hundread millionth time it did not happen before, but I guess I'm just making that up as well.

Geofferey, 2020-11-06:
Working on a possible solution to this. If the code works I'll share with you guys. I think it will be beneficial for everyone and shouldn't break things for others.

closed 2020-11-06 by Geofferey