kismetwireless/kismet #318: websockets 3.0.2 not presend

Haasie, 2020-10-27:
Hello, 
For a little project i try to install kismet on my raspberry pi 3b with dietpi 32bit on it.
from packages all works fine. i want to let it autostart since i want to place the raspberry at some of my clients to monitor there network. 

when i try to compile from source i keep getting this message. 
`configure: error: Package requirements (libwebsockets >= 3.2.0) were not met:

Requested 'libwebsockets >= 3.2.0' but version of libwebsockets is 2.0.3

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libwebsockets_CFLAGS
and libwebsockets_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
`
i have searched already but can't find anything on the internet. 

I hope someone have the solution for me.

With kind regards

kismetwireless, 2020-10-27:
You need a more modern version of websockets from your distribution, or you
news to manually compile a more modern version, or you need to disable
websockets in kismet via configure.

Kismet won't work with a version of libwebsockets that old which is why it
checks the version and fails.

Haasie, 2020-10-27:
I understand that,
i can't find a newer version of libwebsockets for my distro atm. 
google also won't help me with my search

Haasie, 2020-10-27:
otherwise, maby i need to install the package and accept's it run's as root?

dragorn, 2020-10-27:
I'm not familiar with your distribution so you'd have to contact them to
update their packages of build your own libwebsockets. 3.2 is a few years
old at this point.

Geofferey, 2020-10-28:
Maybe try building for yourself like this...
    
    sudo apt-get install libssl-dev
    git clone https://github.com/warmcat/libwebsockets.git
    cd libwebsockets
    git checkout v3.2-stable
    mkdir build
    cd build
    cmake ..
    make
    sudo make install
    sudo ldconfig

Haasie, 2020-10-29:
Hello all,

Thank you for your kind help. 
With the help of a friend of mine we fixed it. 
Here is the solution that worked for me.

`First:

http://ftp.nz.debian.org/debian/pool/main/libw/libwebsockets/libwebsockets17_4.1.3-1_armhf.deb

dpkg -i llibwebsockets17_4.1.3-1_armhf.deb

second:

http://ftp.nz.debian.org/debian/pool/main/libw/libwebsockets/libwebsockets-dev_4.1.3-1_armhf.deb

dpkg -i libwebsockets-dev_4.1.3-1_armhf.deb`

reboot, and all is finished.

Thanks again!

closed 2020-10-29 by Haasie