kismetwireless/kismet #321: Only log access points & SSID filter

AnBrSolutions, 2020-11-09:
Hello,

I must admit that I just started working with KISMET, so please excuse if I missed a certain functionality which already serves my needs. 

I have the following requests which may already be available:
1) Preferably I would like to only log an entry in the kismet db if a specific SSID was found. Reason is that I am looking for a specific SSID in the nearby area. Is this somehow possible?

2) If 1) is not possible, is there a way to only log APs? I would like to reduce the logging to a very minimum and thus would like to not log any clients.

Thanks and best regards

pahanco, 2020-12-16:
Did you read about device and packet filtering [here](https://www.kismetwireless.net/docs/readme/logging/) ? However there isn't filtering by SSID-string, only by MAC, IMHO.

solsticedhiver, 2022-02-11:
The point 2. is adressed with the new wardriving mode in 2022_01_R1 release. Look at https://kismetwireless.net/docs/readme/wardriving/

With that mode, you can only log APs

I don't know much for filter, but I would log all APs wiht the wardriving mode, and then you have multiple possibilities to folter APs afterward.