kismetwireless/kismet #327: API Token with Role readonly usage resulting in Permission Denied

notaco, 2020-12-04:
I am getting 401 Permission denied using readonly keys on endpoints listed as Role readonly in docs (ie /system/status & timestamp) where an admin key is working. I am using a build from the 2020-12-R1 source tar ball. The readonly key was generated via webui drop down to ensure the role is correct. Let me know if more info (ie config) is needed. Not sure if I am misunderstanding and botching the usage, but attaching log of demonstration using curl. Thanks!

`$ curl http://kismet:PASSWORD@127.0.0.1:2501/auth/apikey/list.prettyjson

 [
 
  {
  "kismet.httpd.auth.role": "",
  "kismet.httpd.auth.name": "",
  "kismet.httpd.auth.token": "5D29AC0AD454BEA871F80B4D76DDF271",
  "kismet.httpd.auth.expiration": 0
  },
 
  {
  "kismet.httpd.auth.role": "admin",
  "kismet.httpd.auth.name": "1toke",
  "kismet.httpd.auth.token": "CF6BDC790349B0CC33FA43A2134F99CD",
  "kismet.httpd.auth.expiration": 0
  },
 
  {
  "kismet.httpd.auth.role": "readonly",
  "kismet.httpd.auth.name": "2toke",
  "kismet.httpd.auth.token": "45BDFD3E0B3BFAB5004F96CDB6A16B28",
  "kismet.httpd.auth.expiration": 0
  }
 ]$ curl --cookie "KISMET=CF6BDC790349B0CC33FA43A2134F99CD" -d 'json={"fields": ["kismet.system.version", "kismet.system.git"]}' http://127.0.0.1:2501/system/status.prettyjson

 {
 "description.kismet.system.git": "string, Git commit string",
 "kismet.system.git": "non-git-release",

 "description.kismet.system.version": "string, Kismet version string",
 "kismet.system.version": "2020-12-R1"
 }$ curl --cookie "KISMET=45BDFD3E0B3BFAB5004F96CDB6A16B28" -d 'json={"fields": ["kismet.system.version", "kismet.system.git"]}' http://127.0.0.1:2501/system/status.prettyjson
<html><head><title>401 Permission denied</title></head><body><h1>401 Permission denied</h1><br><p>This resource requires a login or session token.</p></body></html>
$ curl --cookie "KISMET=45BDFD3E0B3BFAB5004F96CDB6A16B28" http://127.0.0.1:2501/system/timestamp.json
<html><head><title>401 Permission denied</title></head><body><h1>401 Permission denied</h1><br><p>This resource requires a login or session token.</p></body></html>
`

kismetwireless, 2020-12-04:
Looks like those actually require a logon role.  I'll change the docs and they'll change a RO role in some future release.

closed 2020-12-04 by kismetwireless

kismetwireless, 2020-12-04:
Updated the docs to show an errata on 2020-12-R1 and updated git to make them the RO role properly; so whenever R2 happens it'll pick up that fix.  Good catch!