kismetwireless/kismet #328: kismet_cap_sdr_rtladsb --tcp mode does not connect

CharlesCage, 2020-12-04:
When I updated to the latest version, all my remote captures just required the --tcp flad to be functional. However, the adsb remote capture doesn't seem to function with --tcp enabled. 

With the following command:
`kismet_cap_sdr_rtladsb --connect xx.xx.xx.xx:3501 --source=rtladsb-11:name=sensors-rtladsb --tcp`

I receive the following:
```
Connecting to remote server xx.xx.xx.xx:3501
True
opening websocket
Failed to connect to remote host:  Malformed HTTP message
Failed to connect to remote host, exiting.
Registering remote source rtladsb rtladsb-11:name=sensors-rtladsb
```
And I see the following in my Kismet web instance:
```
Dec 04 2020 16:09:07 Data source 'sensors-rtladsb / rtladsb-11:name=sensors-rtladsb' ('rtladsb-11') encountered an error: TCP connection error: Connection reset by peer
Dec 04 2020 16:09:07 (DST SETUP REMOTE ERROR) Invalid signature on command frame
```

AM I doing something wrong, or is there an issue with ADSB?

Thanks so much for all your work. Big fan!

kismetwireless, 2020-12-04:
Yep; that's a bug in the python external API library.  Fixing it in Git now...

kismetwireless, 2020-12-04:
Fixed in git, I'll probably fire off a R2 to get this and the RO roles fix out there before the weekend; packages will show up later tonight.

closed 2020-12-04 by kismetwireless

CharlesCage, 2020-12-04:
Thanks so much. You rock.

commit 98df7920 (2020-12-04): Fix handling the --tcp flag in python remote cap tools

Fix handling the --tcp flag in python remote cap tools
Fixes #328