kismetwireless/kismet #330: API `Alerts - device presence / absence - viewingPermalink` NOT WORK

yaphone, 2020-12-10:
hi,
    I have been used Kismet for moths,  recently I updated my Kismet version to `Kismet 2020-12-R3`, as its doc describe, it has a new api  [Alerts - device presence / absence - viewingPermalink ](https://www.kismetwireless.net/docs/devel/webui_rest/devices/#alerts---device-presence--absence---viewing), but I found it NOT work as expect, when send a GET request, it returns

`<html><head><title>404 not found</title></head><body><h1>404 Not Found</h1><br><p>Could not find <code>&#x2F;devices&#x2F;alerts&#x2F;mac&#x2F;found&#x2F;list.json</code></p></body></html>`

can anyone help? Thank you very much!

kismetwireless, 2020-12-10:
Docs updated; should be 'macs.json'

closed 2020-12-10 by kismetwireless

yaphone, 2020-12-11:
THX a lot, it works well !!