kismetwireless/kismet #333: Relative (negative) timestamps not working with '/devices/last-time/[timestamp]/devices.json'

mainakae, 2020-12-30:
Tested on a armhf, 2020-12-R3a (on a Raspberry Pi 4). The non negative version works just fine, but when setting relative timestamps, it returns always all the devices (from the beginning of the scanning).

Thanks for the great work, congratulations and happy christmas!

kismetwireless, 2021-01-04:
Which endpoint are you trying to use?  I can't recreate it so far, here:

curl http://kismet:kismet@localhost:2501/device/views/all/last-time/-10/devices.itjson

for instance returns the expected number of devices.

Is there any chance you have an odd keyboard/language setting where your '-' in the URI isn't a standard 0x2D ascii '-'?

kismetwireless, 2021-01-04:
Oh; i think i spot it in the original /devices/last-time/ endpoint.

commit 91f9c9eb (2021-01-04): Handle negative timestamps in /devices/last-time endpoint

Handle negative timestamps in /devices/last-time endpoint.
Fixes #333
closed 2021-01-04 by kismetwireless