kismetwireless/kismet #339: Python clients hanging since 2020-12

kilspi, 2021-01-28:
Since 2020-12 (last check was master commit da08793d342b9c47a9c34c6921338324c212e19b), python clients are hanging in REST-Calls. Calls done are kismet_rest.Datasources.all() and kismet_rest.Devices.by_mac(), not more than twice per minute, but there 2-3 of this clients running. With 2020-09 the clients run for weeks without problems (kismet server is restarted every night), but since 2020-12 they get stuck within hours. Callstack shows they are definitely hung in socket calls.  

System is Debian bullseye, Kismet compiled from sources.

kismetwireless, 2021-01-28:
Try todays git, or tomorrows nightly packages.  Tweaked the http 1.0/1.1 connection header handling.

kilspi, 2021-01-29:
Looked very promising, but after about 8 hours on of the clients hung again (stopped him 1.5 hours later with CTRL-C):

Traceback (most recent call last):
  File "/home/chris/devcheck/./devcheck.py", line 333, in <module>
    main()
  File "/home/chris/devcheck/./devcheck.py", line 72, in main
    querydevices()
  File "/home/chris/devcheck/./devcheck.py", line 173, in querydevices
    updatetrackingdatasources()
  File "/home/chris/devcheck/./devcheck.py", line 179, in updatetrackingdatasources
    for datasource in _DATASOURCES_REST.all():
  File "/home/chris/.local/lib/python3.9/site-packages/kismet_rest/datasources.py", line 30, in all
    for result in self.interact_yield("GET", url, **callback_settings):
  File "/home/chris/.local/lib/python3.9/site-packages/kismet_rest/base_interface.py", line 256, in interact_yield
    for result in self.process_response_stream(response, **kwargs):
  File "/home/chris/.local/lib/python3.9/site-packages/kismet_rest/base_interface.py", line 272, in process_response_stream
    for result in response.iter_lines(decode_unicode=True):
  File "/usr/lib/python3/dist-packages/requests/models.py", line 797, in iter_lines
    for chunk in self.iter_content(chunk_size=chunk_size, decode_unicode=decode_unicode):
  File "/usr/lib/python3/dist-packages/requests/utils.py", line 520, in stream_decode_response_unicode
    for chunk in iterator:
  File "/usr/lib/python3/dist-packages/requests/models.py", line 753, in generate
    for chunk in self.raw.stream(chunk_size, decode_content=True):
  File "/usr/lib/python3/dist-packages/urllib3/response.py", line 572, in stream
    for line in self.read_chunked(amt, decode_content=decode_content):
  File "/usr/lib/python3/dist-packages/urllib3/response.py", line 764, in read_chunked
    self._update_chunk_length()
  File "/usr/lib/python3/dist-packages/urllib3/response.py", line 694, in _update_chunk_length
    line = self._fp.fp.readline()
  File "/usr/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
KeyboardInterrupt

Another other client is still running well. Both are on the same machine as kismet server, so there should be no network issues.

kilspi, 2021-01-29:
2 hours later:
Traceback (most recent call last):
  File "/home/chris/devcheck/./devcheck.py", line 333, in <module>
    if __name__ == "__main__":
  File "/home/chris/devcheck/./devcheck.py", line 72, in main
    querydevices()
  File "/home/chris/devcheck/./devcheck.py", line 173, in querydevices
    updatehapresence(cleandevname, not isaway)
  File "/home/chris/devcheck/./devcheck.py", line 179, in updatetrackingdatasources
    datasources = dict()
  File "/home/chris/.local/lib/python3.9/site-packages/kismet_rest/datasources.py", line 30, in all
    for result in self.interact_yield("GET", url, **callback_settings):
  File "/home/chris/.local/lib/python3.9/site-packages/kismet_rest/base_interface.py", line 256, in interact_yield
    for result in self.process_response_stream(response, **kwargs):
  File "/home/chris/.local/lib/python3.9/site-packages/kismet_rest/base_interface.py", line 272, in process_response_stream
    for result in response.iter_lines(decode_unicode=True):
  File "/usr/lib/python3/dist-packages/requests/models.py", line 797, in iter_lines
    for chunk in self.iter_content(chunk_size=chunk_size, decode_unicode=decode_unicode):
  File "/usr/lib/python3/dist-packages/requests/utils.py", line 520, in stream_decode_response_unicode
    for chunk in iterator:
  File "/usr/lib/python3/dist-packages/requests/models.py", line 753, in generate
    for chunk in self.raw.stream(chunk_size, decode_content=True):
  File "/usr/lib/python3/dist-packages/urllib3/response.py", line 572, in stream
    for line in self.read_chunked(amt, decode_content=decode_content):
  File "/usr/lib/python3/dist-packages/urllib3/response.py", line 764, in read_chunked
    self._update_chunk_length()
  File "/usr/lib/python3/dist-packages/urllib3/response.py", line 694, in _update_chunk_length
    line = self._fp.fp.readline()
  File "/usr/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
KeyboardInterrupt

kilspi, 2021-02-04:
After some days of testeing, hanging seems much less frequent than before fixing connection handling. But here is another one, this time from querydevices, after about 16 hours of running:

Traceback (most recent call last):
  File "/home/chris/devcheck/devcheck.py", line 335, in <module>
    main()
  File "/home/chris/devcheck/devcheck.py", line 72, in main
    querydevices()
  File "/home/chris/devcheck/devcheck.py", line 88, in querydevices
    for dev in devs:
  File "/home/chris/devcheck/venv/lib/python3.9/site-packages/kismet_rest/devices.py", line 62, in by_mac
    for result in self.interact_yield("POST", url, **call_settings):
  File "/home/chris/devcheck/venv/lib/python3.9/site-packages/kismet_rest/base_interface.py", line 256, in interact_yield
    for result in self.process_response_stream(response, **kwargs):
  File "/home/chris/devcheck/venv/lib/python3.9/site-packages/kismet_rest/base_interface.py", line 272, in process_response_stream
    for result in response.iter_lines(decode_unicode=True):
  File "/home/chris/devcheck/venv/lib/python3.9/site-packages/requests/models.py", line 797, in iter_lines
    for chunk in self.iter_content(chunk_size=chunk_size, decode_unicode=decode_unicode):
  File "/home/chris/devcheck/venv/lib/python3.9/site-packages/requests/utils.py", line 520, in stream_decode_response_unicode
    for chunk in iterator:
  File "/home/chris/devcheck/venv/lib/python3.9/site-packages/requests/models.py", line 753, in generate
    for chunk in self.raw.stream(chunk_size, decode_content=True):
  File "/home/chris/devcheck/venv/lib/python3.9/site-packages/urllib3/response.py", line 572, in stream
    for line in self.read_chunked(amt, decode_content=decode_content):
  File "/home/chris/devcheck/venv/lib/python3.9/site-packages/urllib3/response.py", line 764, in read_chunked
    self._update_chunk_length()
  File "/home/chris/devcheck/venv/lib/python3.9/site-packages/urllib3/response.py", line 694, in _update_chunk_length
    line = self._fp.fp.readline()
  File "/usr/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
KeyboardInterrupt

kilspi, 2021-02-10:
quickly hacked in a timeout parameter into python class, like stated in https://github.com/kismetwireless/python-kismet-rest/issues/10

Still got an timeout, but script continued after that:

90RD2 0210-094329   1 -87      2      0 
90RD2 0210-094610   1 -91      1      0 
90RD2 0210-094756   1 -91      1      0 
90RD2 0210-094756   1 -91      0      0  AW
ERROR 0210-095726 HTTPConnectionPool(host='server2', port=2501): Read timed out.
90RD2 0210-100206   1 -91      0      0  BA
90RD2 0210-100211   1 -86     14      0  BC
90RD2 0210-100221   1 -86      1    120  DF:4A:15

RFDeck, 2026-11-07:
The HTTP server has changed a lot since this was reported. The connection header handling was fixed in 6e7d7d29
right after the report. Since 2026-10-R1, the server also times out requests that stall and clients that stop reading,
after 30 seconds each (21c1107f, 8d1116ab). Master also limits open connections overall and per client address
(7f5c861e). Does this still happen with 2026-10-R1? Setting a timeout on the client side is a good idea either way. If
it still hangs, the endpoint and the time of the hang would help.