kismetwireless/kismet #342: Realtime monitoring does not provide current device channel

omnidepp, 2021-02-04:
Hi all,
I'm monitoring devices through a websocket like this:

```
ws://IPADDRESS/devices/monitor.ws?user=...

{"monitor": "AA:BB:CC:DD:EE:FF", "request": 12345, "rate": 1, "fields": ["kismet.device.base.key", "kismet.device.base.last_time", "kismet.device.base.signal/kismet.common.signal.last_signal", "'kismet.device.base.channel"]}
```

The goal is to get noticed when a device changes the channel or when the signal changes.

However, I only get this:
```
11:23:29 📥 [RECEIVED]: {"'kismet.device.base.channel": 0,"kismet.common.signal.last_signal": -26,"kismet.device.base.last_time": 1612434181,"kismet.device.base.key": "4202770D00000000_B02333314EA4"}
@ 11:23:30 📥 [RECEIVED]: {"'kismet.device.base.channel": 0,"kismet.common.signal.last_signal": -26,"kismet.device.base.last_time": 1612434181,"kismet.device.base.key": "4202770D00000000_B02333314EA4"}
```

The device channel is always shown as 0, which is incorrect.

Any insights?

kismetwireless, 2021-02-04:
{"monitor": "AA:BB:CC:DD:EE:FF", "request": 12345, "rate": 1, "fields": ["kismet.device.base.key", "kismet.device.base.last_time", "kismet.device.base.signal/kismet.common.signal.last_signal", "'kismet.device.base.channel"]}

You have a stray single quote at the beginning of your channel field, which the server is filling in with an int 0 since it can't find a field with that name.

-m

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

omnidepp, 2021-02-05:
Oh dear, that sure fixes it! Sorry about opening an issue!

closed 2021-02-05 by omnidepp