kismetwireless/kismet #344: SOURCEERROR Source wlp1s0u1u1mon has encountered an error. Kismet

michaelmohamed, 2021-03-03:
```
[Logs]    [3/3/2021, 1:52:14 PM] [kismet_signals_collector] ERROR: All configured channels are in error state!
[Logs]    [3/3/2021, 1:52:14 PM] [kismet_signals_collector] ERROR: Data source 'wlp1s0u1u1mon / wlp1s0u1u1mon' ('wlp1s0u1u1mon') encountered an error: Fatal error from remote source
[Logs]    [3/3/2021, 1:52:14 PM] [kismet_signals_collector] ALERT: SOURCEERROR Source wlp1s0u1u1mon has encountered an error. Kismet will attempt to re-open the source in 5 seconds.  (1 failures)
[Logs]    [3/3/2021, 1:52:14 PM] [kismet_signals_collector] ERROR: Source wlp1s0u1u1mon has encountered an error. Kismet will attempt to re-open the source in 5 seconds.  (1 failures)
[Logs]    [3/3/2021, 1:52:19 PM] [kismet_signals_collector] INFO: Attempting to re-open source wlp1s0u1u1mon
[Logs]    [3/3/2021, 1:52:19 PM] [kismet_signals_collector] INFO: wlp1s0u1u1mon could not connect to NetworkManager, cannot automatically prevent interface 'wlp1s0u1u1mon' from being modified if NetworkManager is running: Could not connect: No such file or directory
[Logs]    [3/3/2021, 1:52:19 PM] [kismet_signals_collector] INFO: wlp1s0u1u1mon interface 'wlp1s0u1u1mon' is already in monitor mode
[Logs]    [3/3/2021, 1:52:19 PM] [kismet_signals_collector] INFO: wlp1s0u1u1mon finished configuring wlp1s0u1u1mon, ready to capture
[Logs]    [3/3/2021, 1:52:19 PM] [kismet_signals_collector] ALERT: SOURCEOPEN Source wlp1s0u1u1mon successfully re-opened
c[Logs]    [3/3/2021, 1:52:23 PM] [kismet_signals_collector] ERROR: All configured channels are in error state!
[Logs]    [3/3/2021, 1:52:23 PM] [kismet_signals_collector] ERROR: Data source 'wlp1s0u1u1mon / wlp1s0u1u1mon' ('wlp1s0u1u1mon') encountered an error: Fatal error from remote source
[Logs]    [3/3/2021, 1:52:23 PM] [kismet_signals_collector] ALERT: SOURCEERROR Source wlp1s0u1u1mon has encountered an error. Kismet will attempt to re-open the source in 5 seconds.  (1 failures)
[Logs]    [3/3/2021, 1:52:23 PM] [kismet_signals_collector] ERROR: Source wlp1s0u1u1mon has encountered an error. Kismet will attempt to re-open the source in 5 seconds.  (1 failures)
```

I am using an external wifi adapter. I get this issue after a few minutes of running 2020-12-R3.

If I use 2016-07-R1, I have no problem.

dragorn, 2021-03-03:
What OS? What kernel? What is the card type? Did you upgrade the remote sources as well?

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

michaelmohamed, 2021-03-03:
Thanks for the quick response.

Hi, I am on a Raspberry Pi 4, Kernel 5.4.83-v8, using Kali Linux inside of a docker container.

I am not using any remote sources. 

My source is set as `source= wlp1s0u1u1mon`

michaelmohamed, 2021-03-03:
I may have resolved the issue. I will report back and close if all is OK.

Before I started Kismet, I would run 

```
airmon-ng stop ${WIFI_SOURCE}mon
airmon-ng start $WIFI_SOURCE
kismet --no-ncurses -c wlp1s0u1u1mon
```

When I changed to:

```
kismet --no-ncurses -c wlp1s0u1u1
```

It seems to work. Not sure what is happening here but allowing kismet to set the monitoring mode, things seem OK.

kismetwireless, 2021-03-04:
Nexmon needs specific configuration or it won't make proper monitor interfaces, correct. I would definitely let kismet do it, unless you're sure you're doing the specific nexmon commands.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

michaelmohamed, 2021-03-04:
It is very odd. This error is happening once every hour or so.

kismetwireless, 2021-03-04:
Sounds like "normal" nexmon behavior to me, it's not a very stable patch in monitor mode and is highly dependent on a lot of other factors.

That error means the driver stopped accepting channel set commands; either the driver crashed out, the card stopped responding and the driver crashed out, or something else on your system took over the interface and took it out of monitor mode so it no longer accepts channel set commands (ie, put another vif in managed mode or something like that).

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

michaelmohamed, 2021-03-04:
Thanks for the insight! This was very helpful.

Much appreciated @kismetwireless @dragorn

closed 2021-03-04 by michaelmohamed

michaelmohamed, 2021-03-04:
It seems that the problem was happening about every 7 minutes. I was able to figure out that the NetworkManager may have been trying to take control. 

```
# Showed that wlp1s0u1u1 as "disconnected"
nmcli dev status

# Changed the device to be unmanaged (solution)
nmcli device set wlp1s0u1u1 managed no
```

michaelmohamed, 2021-03-04:
This is likely related to the kismet output:

```
[Logs]    [3/4/2021, 10:34:57 AM] [] INFO: wlp1s0u1u1 could not connect to NetworkManager, cannot automatically prevent interface 'wlp1s0u1u1' from being modified if NetworkManager is running: Could not connect: No such file or directory
```

michaelmohamed, 2021-03-04:
The ultimate solution was to add `ENV DBUS_SYSTEM_BUS_ADDRESS=unix:path=/host/run/dbus/system_bus_socket` to my docker file.