kismetwireless/kismet #347: Is there a topic to subscribe to new device data?

michaelmohamed, 2021-03-05:
https://kismetwireless.net/docs/devel/webui_rest/eventbus/

I noticed the above documentation. Is there a topic for device data, similar to what is returned by the rest API? 

Thank You.

  M

michaelmohamed-anno, 2021-03-06:
https://www.kismetwireless.net/docs/devel/webui_rest/devices/#realtime-device-monitoring

michaelmohamed-anno, 2021-03-06:
From the example, when I use:

 websocat ws://user:pass@localhost:2501/devices/monitor.ws
  {"monitor": "AA:BB:CC:00:00:00/FF:FF:FF:00:00:00", "request": 31337, "rate": 1, "fields": ["kismet.device.base.key", "kismet.device.base.last_time", "kismet.device.base.signal/kismet.common.signal.last_signal"]}

I see the following in my kismet logs:

```
[Logs]    [3/6/2021, 12:05:21 PM] [] ERROR: Invalid device monitor request: * Line 2, Column 1
[Logs]    [3/6/2021, 12:05:21 PM] []   Syntax error: value, object or array expected.
```

Any insight into this?

michaelmohamed-anno, 2021-03-06:
When I use the following, the cli just hangs, but in my kistmet logs, I see device detections

```
> websocat 'ws://192.168.86.34:2501/devices/monitor.ws?user=kismet&password=kismet'
{"monitor": "*", "request": 31337, "rate": 1}
```

michaelmohamed, 2021-03-06:
I think that I found what I am looking for here:

https://www.kismetwireless.net/docs/devel/webui_rest/devices/#realtime-device-monitoring

When I use the following, the cli just waits (no output), but in my kistmet logs, I see device detections

Am I missing something? Thanks for your help.

```
> websocat 'ws://192.168.86.34:2501/devices/monitor.ws?user=kismet&password=kismet'
{"monitor": "*", "request": 31337, "rate": 1}
```

mm-e1, 2021-03-30:
Seems to work in master. Closing.

closed 2021-03-30 by mm-e1