kismetwireless/kismet #354: Difference between a Wi-Fi “Device” and a Wi-Fi “Client”

cup-of-coffee-m8, 2021-03-15:
It's been looking at the JSON data output from Kismet, and I can't figure-out the attributes that separate a sniffed Wi-Fi "Client" and "Device".

Originally I thought that clients were distinguished by being associated with an AP (shown by the dot11_device_last_basis field), however, there are both clients and devices which have applicable entries - can anyone help me out!? 

Thanks in advance!

dragorn, 2021-03-15:
A device is something that has been detected on a wi-fi data source, but no further information is available about it.

A client is something that has been detected on a wi-fi data source, and has been seen to act like a client, such as being the source of a to-ds packet, sending a management frame like a probe request, etc - something we know for sure is acting like a wi-fi client.

-m

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

cup-of-coffee-m8, 2021-03-15:
Thank you! And thank you for your quick response! So in that case, how is the dot11_last_bssid field populated? I would have thought that you'd only be able to do this through either association packets, or managements packets?

I only ask, because I have the JSON data open side-by-side of both a client and a device, and the only difference I can see is that the client has the embedded "kismet_device_base_datasize_rrd" struct whereas the client does not. That appears to be the only thing that differentiates the two. Is there anything else that could possibly affect this?

dragorn, 2021-03-15:
(almost) every wifi packet has 3 addresses - source, dest, and bssid. So we know it's related to a given network, but we may not know if it's a wifi client or a bridged device, for example if there was traffic *to* it, but we haven't seen anything back from it yet. Could be a wired device bridged to wifi, could be a client we haven't seen.

src != bssid, fromds = we know it's bridged. it might be a wifi client elsewhere, but for this bssid, it's bridged.
tods = we know it's a wifi client since it's sending into the network

But dest != bssid, fromds? we know something is getting talked at, but we don't know what kind of device it is yet.

-m

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

kismetwireless, 2021-03-15:
And the datasize rrd is a round-robin history of data seen - if no data has been seen, that won't exist, to save memory.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

cup-of-coffee-m8, 2021-03-15:
Thank you both for taking the time to explain that, it makes sense now! Stay safe!