kismetwireless/kismet #357: SOURCEERROR Source ADSB has encountered an error. Kismet will attempt to re-open the source in 5 seconds

mm-e1, 2021-03-19:
I have built from master, and also used the latest release from github. I am using the source: `source=rtladsb-0:name=ADSB`

I am often running into the error:

```
SOURCEERROR Source ADSB has encountered an error. Kismet will attempt to re-open the source in 5 seconds
```
It seems that when there is a source error, the app is unable to re-connect to the SDR successfully.

This does not seem to happen for TPMS. 

This may be related to: https://github.com/kismetwireless/kismet/issues/181

I am able to reproduce by running kismet. If I shake the SDR/USB a bit, the error happens, and then it seems to not be able to reconnect.

dragorn, 2021-03-19:
Check dmesg, lsusb, etc. Often a rtlsdr will just go out to lunch until unplugged and replugged; the hw isn't particularly reliable in most cases. There may be errors in dmesg.

If unplugging and plugging the rtlsdr back in fixes it, then it's a hw problem, unfortunately.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

mm-e1, 2021-03-19:
Interesting. I have tried 6 nooelec v4. 

I don't run into this problem if I run rtl_adsb manually.

I'll give another brand a try as well

mm-e1, 2021-03-19:
Unplugging and plugging it back in does not fix the problem.

mm-e1, 2021-03-19:
I'm getting mixed results. 

Using:

source=rtl433-0:channel=315MHz,gain=60,name=TPMS315

If I unplug the usb rtl, I get no error. When I plug it back in, I see no prompt or update to the logs.

Using 

source=rtladsb-0:name=ADSB

If I unplug the usb rtl, I get and error that it is reconnecting. When I plug it in, it does not reconnect and I keep getting an error.

kismetwireless, 2021-03-19:
Unfortunately I have multiple adsb and 433 sources running for weeks at a time with no issue, so if you're able to get any other sort of info from the system about why the source is terminating, that would be helpful.

For the adsb source to terminate, something is killing the process, or librtlsdr is returning an error.

The rtl433 source operates completely differently, and counts on the rtl433 binary returning an error or exiting to detect if the radio has been disconnected; if the rtl433 tool doesnt exit and implements its own health checks and waiting, kismet won't have any idea.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

mm-e1, 2021-03-19:
Thanks for the info. This is specific to if you unplug and then plug the sdr back in while kismet is running. There seems to be some sort of continuous loop. 

Attempting to re-open source ADSB
SOURCEOPEN Source ADSB successfully re-opened
SOURCEERROR Source ADSB has encountered an error. Kismet will attempt to re-open the source in 5 seconds. (1 failures)

indefinitely. does this happen to you, if you unplug and re-plug your sdr?

Much appreciated. Thanks.

```
Mar 19 2021 17:15:02 Attempting to re-open source ADSB
Mar 19 2021 17:14:57 Source ADSB has encountered an error. Kismet will attempt to re-open the source in 5 seconds. (1 failures)
Mar 19 2021 17:14:57 SOURCEERROR Source ADSB has encountered an error. Kismet will attempt to re-open the source in 5 seconds. (1 failures)
Mar 19 2021 17:14:57 Data source 'ADSB / rtladsb-0:name=ADSB' ('rtladsb-0') encountered an error: IPC connection closed
Mar 19 2021 17:14:57 SOURCEOPEN Source ADSB successfully re-opened
Mar 19 2021 17:14:56 Attempting to re-open source ADSB
Mar 19 2021 17:14:51 Source ADSB has encountered an error. Kismet will attempt to re-open the source in 5 seconds. (1 failures)
Mar 19 2021 17:14:51 SOURCEERROR Source ADSB has encountered an error. Kismet will attempt to re-open the source in 5 seconds. (1 failures)
Mar 19 2021 17:14:51 Data source 'ADSB / rtladsb-0:name=ADSB' ('rtladsb-0') encountered an error: IPC connection closed
Mar 19 2021 17:14:51 SOURCEOPEN Source ADSB successfully re-opened
Mar 19 2021 17:14:50 Attempting to re-open source ADSB
Mar 19 2021 17:14:45 Source ADSB has encountered an error. Kismet will attempt to re-open the source in 5 seconds. (1 failures)
Mar 19 2021 17:14:45 SOURCEERROR Source ADSB has encountered an error. Kismet will attempt to re-open the source in 5 seconds. (1 failures)
Mar 19 2021 17:14:45 Data source 'ADSB / rtladsb-0:name=ADSB' ('rtladsb-0') encountered an error: IPC connection closed
Mar 19 2021 17:14:45 SOURCEOPEN Source ADSB successfully re-opened
```

yesimxev, 2023-12-27:
You have to disable allocating zerocopy buffers. It helps with rtl_adsb but I couldn't figure out with kismet source yet