kismetwireless/kismet #361: Kismet Package install missing data

gomker, 2021-03-28:
After installing via `apt install kismet` I don't have a executable in `/usr/bin`

```
Package: kismet
Version: 2020-12-R3a
Priority: extra
Section: default
Maintainer: <root@7d0d9bdce676>
Installed-Size: unknown
Depends: kismet-core, kismet-adsb-icao-data, kismet-capture-linux-wifi, kismet-capture-linux-wifi, kismet-capture-linux-bluetooth, kismet-capture-nrf-mousejack, python3-kismetcapturertl433, python3-kismetcapturertladsb, python3-kismetcapturertlamr, python3-kismetcapturefreaklabszigbee, kismet-capture-ti-cc-2531, kismet-capture-ubertooth-one, kismet-capture-nrf-51822, kismet-capture-nxp-kw41z, kismet-logtools
Homepage: http://example.com/no-uri-given
License: unknown
Vendor: none
Download-Size: 1,130 B
APT-Sources: https://www.kismetwireless.net/repos/apt/release/focal focal/main amd64 Packages
Description: Kismet metapackage
```
I've attempted to do a `remove` `purge` `reinstall` so far.  I had the kali package installed before, seeing if that caused some sort of conflict

kismetwireless, 2021-03-28:
the 'kismet' package is a metapackage that includes the others; kismet-core contains the kismet binary; you'll need it and the capture packages for proper functionality.

the kismet-core debs contain the executable as expected so i suspect you've got something screwy in your package management; maybe try removing all of the kismet packages (including all the individual component packages) and re-installing?

-m

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

gomker, 2021-03-28:
Thanks for confirming - I do seem to have 3 sources for kismet from `apt show` I will try and purge.

closed 2021-03-28 by gomker

gomker, 2021-03-28:
For others reference if someone gets the same type of issue

- Check for versions with `sudo apt show kismet -a` - You might have more than one
- install with `sudo apt install kismet=<correct version>`