kismetwireless/kismet #363: How to detect KRACK Attack on Kismet?

fachrurozyr99, 2021-04-01:
I've been configure the NONCEREUSE Alert in Kismet_alert.conf, But when i do some KRACK Attack to the target, theirs no alert Show? 

Help, please

fachrurozyr99, 2021-04-03:
nobody can answer?

kismetwireless, 2021-04-04:
Nonce reuse detection is only possible when 

a) you are on the proper channel at the proper time, 
b) you've seen the original exchange, and 
c) your device is capable of receiving all the packets required in the first place.

Handshakes are performed as data frames.  You will never see a handshake performed at a higher data rate than your hardware is capable of receiving in the first place, and you will miss any handshake happening while you are channel hopping to another channel.  All the normal warnings about missing data frames apply - 802.11n and faster can be extremely difficult to successfully capture data frames from due to mimo and related, chances of your capture card missing handshakes are very high.

It is to the best of my knowledge impossible to differentiate between a krack attack and a normal retransmission as a completely passive observer, which is why that alert is disabled by default.

closed 2021-04-04 by kismetwireless