kismetwireless/kismet #364: Kismet Can't Detect WPS BRUTE FORCE

fachrurozyr99, 2021-04-01:
I do another detection using kismet, i trying to do WPS Brute FORCE with reaver in airgeddon, But My Kismet WIDS cant Detect the Attack and show the alerts. It's Something wrong with my kismet?

Scenario, I got 1 AP As victim, 1 Device as WIDS Kismet, 1 Device as Attack with airgeddon.

Help me if u have same problem and solved it. Please thank you

fachrurozyr99, 2021-04-04:
help

kismetwireless, 2021-04-04:
WPS brute force detection relies on capturing enough packets to consider a brute force to be happening.  If you are monitoring a specific AP you need to disable channel hopping or you will likely miss most of the packets.

closed 2021-04-04 by kismetwireless

fachrurozyr99, 2021-04-05:
So, I need sent more of brutefoce packet to make kismet can detect that as a alert of attack ?

fachrurozyr99, 2021-04-05:
***@***.***D72A1D.B22A6520]

I Have try to attack my AP with WPS Brute force, but kismet just can detect noclientmfp for the alert. Are you know what is the problem sir? Cause I have spent 2 hours to brute force, but no one alert about WPSBrute shown