kismetwireless/kismet #365: Webserver not not accessible

VasAthanasios, 2021-04-01:
Hi,

I compiled master on Ubuntu Focal 20.04.2 LTS, and the localhost:2501 does not appear. 

Running _kismet -c wlan0_ does work and I get new detections on the terminal. I can run some old python code I wrote about 6 month ago to extract data through the REST and it works fine, its just the webserver that does not appear.

After looking over the documentation I found out that the "~/.kismet/kismet_httpd.conf" is not there. I am compiling -b kismet-2020-12 to check if issue is resolved. I will report any findings.

Thank you for your time.

kismetwireless, 2021-04-04:
Kismet cannot start, at all, without launching the http server, so it's likely either you're not running the kismet version you think you are (if it does not have a banner in the terminal telling you to go to the webserver, it's likely an old version from another package you installed somewhere) or you've got some sort of firewall or host network configuration option, check ufw and that your loopback interface is configured, try by IP instead of hostname, etc.

If you're mixing configs from multiple versions I suggest just deleting all of them and installing a clean setup.

VasAthanasios, 2021-04-05:
Thank you very much for your response.

I recreated the process on a VM and it worked with no issue. The issue is probably related to the fact that I am running it on a docker container. 

Before I close this issue, could you please let me know if Kismet is tested / supported on Docker containers?

Thank you for your time.

kismetwireless, 2021-04-05:
Ah yes that would be a critical difference.

You'll have to expose the port via docker if you want to talk to it from outside docker.

People have made kismet run in docker but it's not trivial to expose the network interfaces and control sockets. It's not something I do or have any recipes for, so while you can probably get it to work, you're on your own for the most part.

-------- Original Message --------

closed 2021-04-05 by VasAthanasios