kismetwireless/kismet #366: How to decapsulate traffic with WPA password in Kismet

fachrurozyr99, 2021-04-01:
I need some help how we can decrypt WPA password in Kismet to detect WPA brute Force attack or Krack attack, cause kisme cant detect attack traffic with password

dragorn, 2021-04-01:
Kismet currently does not support decrypting WPA on the fly.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

fachrurozyr99, 2021-04-02:
> Kismet currently does not support decrypting WPA on the fly. ‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐
> […](#)

did u  know how kismet work to detect KRACK attack and WPS Bruteforce? cause kismet already have that alert, but how it's work if cant decrypting the WPA packet during work? thanks

fachrurozyr99, 2021-04-03:
> Kismet currently does not support decrypting WPA on the fly. ‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐
> […](#)

sorry for asking again, if decrypting WPA no on the fly it's possible?

kismetwireless, 2021-04-04:
Kismet currently does not support decrypting WPA on the fly.

closed 2021-04-04 by kismetwireless