kismetwireless/kismet #367: FATAL: Kismet an error occurs when accessing the 127.0.0.1:2501

defencore, 2021-04-08:
HW INFO:
- Qualcomm Atheros QCA9533
- RAM 128MB / SPI 16MB + NAND 128MB

The kismet server crashes with an error when accessing from a web browser to <Kismet_IP>:2501

`ADSB ICAO lookup will not be available.
INFO: Registered PHY handler 'RTLADSB' as ID 8
INFO: Could not open system plugin directory (/usr/lib/kismet/), skipping: 
      No such file or directory
INFO: Did not find a user plugin directory (/root/.kismet//plugins/), 
      skipping: No such file or directory
INFO: GPS track will be logged to the Kismet logfile
ERROR: Error reading config file '/root/.kismet/kismet_httpd.conf': No 
       such file or directory
LOCAL: This is the first time Kismet has been run as this user.  You will 
       need to set an administrator username and password before you can 
       use any features of Kismet.  Visit http://localhost:2501/ to 
       configure the initial login, or consult the Kismet documentation at 
       https://www.kismetwireless.net/docs/readme/webserver/ about how to 
       set a password manually.
ERROR: (HTTPD) Could not read session data file, skipping loading saved 
       sessions.
INFO: Serving static file content from /etc/kismet/http_data
INFO: Enabling channel hopping by default on sources which support channel 
      control.
INFO: Setting default channel hop rate to 5/sec
INFO: Enabling channel list splitting on sources which share the same list 
      of channels
INFO: Enabling channel list shuffling to optimize overlaps
INFO: Sources will be re-opened if they encounter an error
INFO: Saving datasources to the Kismet database log every 30 seconds.
INFO: Launching remote capture server on 127.0.0.1 3501
INFO: No data sources defined; Kismet will not capture anything until a 
      source is added.
INFO: Opened kismetdb log file '/tmp//Kismet-20200729-18-56-22-1.kismet'
INFO: Saving packets to the Kismet database log.
ALERT: rootuser Kismet is running as root; this is less secure.  If you 
       are running Kismet at boot via systemd, make sure to use `systemctl 
       edit kismet.service` to change the user.  For more information, see 
       the Kismet README for setting up Kismet with minimal privileges.
INFO: Starting Kismet web server...
INFO: (DEBUG) Beast server listening on 0.0.0.0:2501
Assertion failed: map_[j][0] == 0 (./boost/beast/http/impl/field.ipp: field_table: 457)
Aborted`

matsura8, 2021-04-18:
Same issue on an X230 running Kali.
Additional info:
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff674c700 (LWP 171618)]
[New Thread 0x7ffff5f4b700 (LWP 171619)]
[New Thread 0x7ffff574a700 (LWP 171620)]
[New Thread 0x7ffff4f49700 (LWP 171621)]
[New Thread 0x7ffff4748700 (LWP 171622)]
[New Thread 0x7ffff3f47700 (LWP 171623)]
[New Thread 0x7ffff3746700 (LWP 171624)]
[New Thread 0x7ffff2f45700 (LWP 171625)]
[New Thread 0x7ffff2744700 (LWP 171626)]
[New Thread 0x7ffff1f43700 (LWP 171627)]
[New Thread 0x7ffff1742700 (LWP 171628)]
[New Thread 0x7ffff0f41700 (LWP 171629)]
[New Thread 0x7ffff0740700 (LWP 171630)]
[New Thread 0x7fffeff3f700 (LWP 171631)]
[New Thread 0x7fffef73e700 (LWP 171632)]
[New Thread 0x7fffeef3d700 (LWP 171633)]
[New Thread 0x7fffee73c700 (LWP 171634)]
INFO: Including sub-config file: /usr/local/etc/kismet_httpd.conf
INFO: Including sub-config file: /usr/local/etc/kismet_memory.conf
INFO: Including sub-config file: /usr/local/etc/kismet_alerts.conf
INFO: Including sub-config file: /usr/local/etc/kismet_80211.conf
INFO: Including sub-config file: /usr/local/etc/kismet_logging.conf
INFO: Including sub-config file: /usr/local/etc/kismet_filter.conf
INFO: Including sub-config file: /usr/local/etc/kismet_uav.conf
INFO: Loading config override file '/usr/local/etc/kismet_package.conf'
INFO: Optional sub-config file not present: /usr/local/etc/kismet_package.conf
INFO: Loading config override file '/usr/local/etc/kismet_site.conf'
INFO: Optional sub-config file not present: /usr/local/etc/kismet_site.conf
INFO: Setting server UUID A643ECA0-744A-11EB-8F77-54454D53494B
INFO: Starting Beast webserver on 0.0.0.0:2501
INFO: Opened OUI file '/usr/local/share/kismet/kismet_manuf.txt.gz
INFO: Indexing manufacturer db
INFO: Completed indexing manufacturer db, 29050 lines 581 indexes
[New Thread 0x7fffedf27700 (LWP 171635)]
[New Thread 0x7fffed726700 (LWP 171636)]
[New Thread 0x7fffecf25700 (LWP 171637)]
[New Thread 0x7fffdffff700 (LWP 171638)]
INFO: Saving devices to the Kismet database log every 30 seconds.
INFO: Using default rates of 10/min, 1/sec for alert 'DEVICEFOUND'
INFO: Using default rates of 10/min, 1/sec for alert 'DEVICELOST'
INFO: Registering support for DLT_PPI packet header decoding
INFO: Registering support for DLT_RADIOTAP packet header decoding
INFO: Registering support for DLT_BTLE_LL_RADIO packet header decoding
INFO: Using default rates of 10/min, 1/sec for alert 'BADFIXLENIE'
INFO: PHY802.11 will not process Wi-Fi 'phy' and 'control' frames; these typically are the most susceptible to corruption resulting in false devices.  This can be re-enabled with dot11_process_phy=true
INFO: PHY80211 will only process AP signal levels from beacons
INFO: Allowing Kismet clients to view WEP keys
INFO: Keeping EAPOL packets in memory for easy download and WIDS functionality; this can use more RAM.
/devices/views/phydot11_accesspoints/last-time/:timestamp/devices
INFO: Registered PHY handler 'IEEE802.11' as ID 0
INFO: Registered PHY handler 'RTL433' as ID 1
INFO: Registered PHY handler 'Z-Wave' as ID 2
INFO: Registered PHY handler 'Bluetooth' as ID 3
INFO: Registered PHY handler 'UAV' as ID 4
INFO: Registered PHY handler 'NrfMousejack' as ID 5
INFO: Using default rates of 10/min, 1/sec for alert 'BLEEDINGTOOTH'
INFO: Registered PHY handler 'BTLE' as ID 6
INFO: Registered PHY handler 'RTLAMR' as ID 7
INFO: Indexing ADSB ICAO db
INFO: Completed indexing ADSB ICAO db, 325208 lines 6505 indexes
INFO: Registered PHY handler 'RTLADSB' as ID 8
INFO: Registered PHY handler '802.15.4' as ID 9
INFO: GPS track will be logged to the Kismet logfile
INFO: Serving static file content from /usr/local/share/kismet/httpd/
INFO: Enabling channel hopping by default on sources which support channel control.
INFO: Setting default channel hop rate to 5/sec
INFO: Enabling channel list splitting on sources which share the same list of channels
INFO: Enabling channel list shuffling to optimize overlaps
INFO: Sources will be re-opened if they encounter an error
INFO: Saving datasources to the Kismet database log every 30 seconds.
INFO: Launching remote capture server on 127.0.0.1 3501
[Thread 0x7fffee73c700 (LWP 171634) exited]
/devices/views/all/last-time/:timestamp/devices

*** KISMET IS SHUTTING DOWN ***

*** KISMET HAS ENCOUNTERED A FATAL ERROR AND CANNOT CONTINUE.  ***
Shutting down plugins...
WARNING: Kismet changes the configuration of network devices.
         In most cases you will need to restart networking for
         your interface (varies per distribution/OS, but 
         typically one of:
         sudo service networking restart
         sudo /etc/init.d/networking restart
         or
         nmcli device set [device] managed true

Kismet exiting.
[Thread 0x7fffeef3d700 (LWP 171633) exited]
[Thread 0x7fffef73e700 (LWP 171632) exited]
[Thread 0x7fffeff3f700 (LWP 171631) exited]
[Thread 0x7ffff0740700 (LWP 171630) exited]
[Thread 0x7ffff0f41700 (LWP 171629) exited]
[Thread 0x7ffff1742700 (LWP 171628) exited]
[Thread 0x7ffff1f43700 (LWP 171627) exited]
[Thread 0x7ffff2744700 (LWP 171626) exited]
[Thread 0x7ffff2f45700 (LWP 171625) exited]
[Thread 0x7ffff3746700 (LWP 171624) exited]
[Thread 0x7ffff3f47700 (LWP 171623) exited]
[Thread 0x7fffed726700 (LWP 171636) exited]
[Thread 0x7fffedf27700 (LWP 171635) exited]
[Thread 0x7fffecf25700 (LWP 171637) exited]
[Thread 0x7ffff4748700 (LWP 171622) exited]
[Thread 0x7ffff4f49700 (LWP 171621) exited]
[Thread 0x7ffff574a700 (LWP 171620) exited]
[Thread 0x7ffff5f4b700 (LWP 171619) exited]
[Thread 0x7ffff674c700 (LWP 171618) exited]
[Thread 0x7ffff70d4880 (LWP 171614) exited]
--Type <RET> for more, q to quit, c to continue without paging--
[Inferior 1 (process 171614) exited with code 01]

defencore, 2021-12-27:
Updating the firmware of the device where the kismet was installed solved the problem.

closed 2021-12-27 by defencore