kismetwireless/kismet #369: external HTTP helper plugins - can not find IPC binary

omnidepp, 2021-04-15:
I'm trying to run the external http helper plugins examples. They're installing just fine and I can manually execute the binaries (i.e. /usr/local/bin/kismet_eventbus)
However, they're not loading when I start kismet - I keep getting these errors:
```
ERROR: Kismet external interface can not find IPC binary for launch:
       kismet_eventbus
ERROR: eventbusproxy failed to launch helper binary 'kismet_eventbus'
```
I cannot find any documentation on where to put these helper binaries (except somewhere on the $PATH). Wouldn't it make more sense in general to be able to specify a full path in the manifest file of the plugin instead of just the name of the helper binary?

dragorn, 2021-04-15:
The binaries are loaded from `helper_binary_path` in the configs.

The path is assembled from that config; if you changed that, or your config paths elsewhere or something, that would cause them to not be found. If you're running in some weird environment where you're trying to map into docker or something you'll need to provide those tools inside that environment. External helpers are loaded the same way as capture drivers and any other external binary tool in kismte.

Having plugin manifests specify exact paths would break any sort of cross-platform or packaging, so I'm not going to allow raw paths in manifests; you could locally patch it in if you wanted but that would break too much that isn't broken for me to want to merge it.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

omnidepp, 2021-04-16:
Thanks for detailed instructions and explanations.
Strangely enough, a default kismet install from package in kali linux or ubuntu requires to add /usr/local/bin to helper_binary_path in kismet.conf. 
Some reference to this line in the config file in the plugin development documentation would be helpful.

closed 2021-04-16 by omnidepp