kismetwireless/kismet #377: Runtime error - Kismetdb_to_Wigglecsv

ottoxgam, 2021-05-22:
Using the latest commit on a Raspberry pi4, running kismetdb_to_wiglecsv aborts with the following error.
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to prepare statement:  FROM data WHERE (lat <> ? AND lon <> ? AND phyname == ? OR phyname == ?) near "FROM": syntax error
Aborted
```

dragorn, 2021-05-22:
It looks like some versions of gcc don't compile the templates properly; in the interim you can try updating to more modern gcc or clang and rebuilding (or even just rebuilding that tool)

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

kismetwireless, 2021-05-23:
Might be fixed in git; if you want to give it a try either building it, or pulling the nightlies tomorrow (they'll trigger building in a few hours)

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

ottoxgam, 2021-05-23:
I'm waiting for GCC 10 to compile to see if that'll help. I'll try the Nightly after that. Thanks.

dragorn, 2021-05-23:
If you don't want to deal with the compiler there should be nightly packages up now with the fix included.

-------- Original Message --------

ottoxgam, 2021-05-23:
Running Nightly kismet/buster,now 9999+git2021-05-23r6c9372b0-1 armhf [installed] and I'm receiving the error still. I'm working on compiling the latest commit now.

dragorn, 2021-05-24:
I think I finally tracked it down, fix should hit tomorrows nightlies & is in git now

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

RFDeck, 2026-11-08:
This looks fixed. The empty `SELECT` came from the Bluetooth field list, which was only filled for some database
versions. 339c8ea5 ("Handle db versions properly in wigle to csv") gives it a default, and 6a0bae2c makes an empty
field list an error instead of bad SQL. Both are in every release since then, including 2026-10-R1.

I exported a current kismetdb (version 12, GPS on) with `kismetdb_to_wiglecsv` from master and it ran without
errors. @ottoxgam, if you still see this with a recent build, please post the log; otherwise this can probably be
closed.