kismetwireless/kismet #380: Getting mon1 could not bring up capture interface 'mon1' on PI4

zekrifon, 2021-06-13:
I compiled and installed Kismet 'master' according to documentation.

`journalctl -u kismet
-- Logs begin at Sun 2021-06-13 18:58:38 EDT, end at Sun 2021-06-13 19:05:18 EDT. --
Jun 13 18:58:44 raspberrypi systemd[1]: Started Kismet.
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Including sub-config file: /usr/local/etc/kismet_httpd.conf
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Including sub-config file: /usr/local/etc/kismet_memory.conf
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Including sub-config file: /usr/local/etc/kismet_alerts.conf
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Including sub-config file: /usr/local/etc/kismet_80211.conf
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Including sub-config file: /usr/local/etc/kismet_logging.conf
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Including sub-config file: /usr/local/etc/kismet_filter.conf
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Including sub-config file: /usr/local/etc/kismet_uav.conf
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Loading config override file '/usr/local/etc/kismet_package.conf'
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Optional sub-config file not present: /usr/local/etc/kismet_package.conf
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Loading config override file '/usr/local/etc/kismet_site.conf'
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Loading optional sub-config file: /usr/local/etc/kismet_site.conf
Jun 13 18:58:46 raspberrypi kismet[672]: INFO: Setting server UUID 9C8493AE-CC7E-11EB-97FB-4B49534D4554
Jun 13 18:58:47 raspberrypi kismet[672]: INFO: Starting Beast webserver on 0.0.0.0:2501
Jun 13 18:58:47 raspberrypi kismet[672]: INFO: Opened OUI file '/usr/local/share/kismet/kismet_manuf.txt.gz
Jun 13 18:58:47 raspberrypi kismet[672]: INFO: Indexing manufacturer db
Jun 13 18:58:47 raspberrypi kismet[672]: INFO: Completed indexing manufacturer db, 30041 lines 601 indexes
Jun 13 18:58:49 raspberrypi kismet[672]: INFO: Launching remote capture server on 127.0.0.1 3501
Jun 13 18:58:49 raspberrypi kismet[672]: INFO: No data sources defined; Kismet will not capture anything until a source is added.
Jun 13 18:58:51 raspberrypi kismet[672]: /devices/views/all/last-time/:timestamp/devices
Jun 13 18:58:51 raspberrypi kismet[672]: INFO: Opened kismetdb log file '/home/pi/.kismet/logs//Kismet-20210613-22-58-46-1.kismet'
Jun 13 18:58:51 raspberrypi kismet[672]: INFO: Saving packets to the Kismet database log.
Jun 13 18:58:51 raspberrypi kismet[672]: INFO: Starting Kismet web server...
Jun 13 18:58:49 raspberrypi kismet[672]: INFO: Launching remote capture server on 127.0.0.1 3501
Jun 13 18:58:49 raspberrypi kismet[672]: INFO: No data sources defined; Kismet will not capture anything until a source is added.
Jun 13 18:58:51 raspberrypi kismet[672]: /devices/views/all/last-time/:timestamp/devices
Jun 13 18:58:51 raspberrypi kismet[672]: INFO: Opened kismetdb log file '/home/pi/.kismet/logs//Kismet-20210613-22-58-46-1.kismet'
Jun 13 18:58:51 raspberrypi kismet[672]: INFO: Saving packets to the Kismet database log.
Jun 13 18:58:51 raspberrypi kismet[672]: INFO: Starting Kismet web server...
Jun 13 18:58:51 raspberrypi kismet[672]: INFO: (DEBUG) Beast server listening on 0.0.0.0:2501
Jun 13 19:02:03 raspberrypi kismet[672]: FATAL:  Datasource helper failed, could not process incoming control packet.
Jun 13 19:02:03 raspberrypi kismet[672]: INFO: mon1 interface 'mon1' is already in monitor mode
Jun 13 19:02:03 raspberrypi kismet[672]: ERROR: Data source 'mon1 / mon1:type=linuxwifi' ('mon1') encountered an error: mon1 could not bring up capture interface 'mon1', check 'dmesg' for possible errors while loading firmware: failed to set flags on interface 'mon1': Operation not permitted
Jun 13 19:02:03 raspberrypi kismet[672]: ALERT: SOURCEERROR Source mon1 (5FE308BD-0000-0000-0000-000F60048CE5) has encountered an error (mon1 could not bring up capture interface 'mon1', check 'dmesg' for possible errors while loading firmware: failed to set flags on interface 'mon1': Operation not permitted) Kismet will attempt to re-open the source in 5 seconds.  (1 failures)
Jun 13 19:02:03 raspberrypi kismet[672]: ERROR: Source mon1 (5FE308BD-0000-0000-0000-000F60048CE5) has encountered an error (mon1 could not bring up capture interface 'mon1', check 'dmesg' for possible errors while loading firmware: failed to set flags on interface 'mon1': Operation not permitted) Kismet will attempt to re-open the source in 5 seconds.  (1 failures)
Jun 13 19:02:03 raspberrypi kismet[672]: ERROR: Kismet external interface got an error writing packet, IPC, TCP, callbacks closed
Jun 13 19:02:08 raspberrypi kismet[672]: INFO: Attempting to re-open source mon1`

dragorn, 2021-06-14:
You didn't provide much context, but:

ERROR: Source mon1 (5FE308BD-0000-0000-0000-000F60048CE5) has encountered an error (mon1 could not bring up capture interface 'mon1', check 'dmesg' for possible errors while loading firmware: failed to set flags on interface 'mon1': Operation not permitted)

Either your user lacks the permissions (and you didn't install suidroot), or you're trying to do monitor mode on a stock rpi internal wifi interface without nexmon patches.

-m

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

zekrifon, 2021-06-14:
> You didn't provide much context, but: ERROR: Source mon1 (5FE308BD-0000-0000-0000-000F60048CE5) has encountered an error (mon1 could not bring up capture interface 'mon1', check 'dmesg' for possible [...]

Hi, thanks for the reply. I'm a real beginner here, so please let me know what could provide more context. 

But, I did install kismet with **sudo make suidinstall**.

And I'm using an external WiFi adapter not the internal RPi WiFi. 

Here's the output of **iw phy phy1 info** :
`pi@raspberrypi:~ $ iw phy phy1 info
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short long limit: 2
        Coverage class: 0 (up to 0m)
        Device supports RSN-IBSS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 1:
                Capabilities: 0x17e
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 32767 bytes (exponent: 0x002)
                Minimum RX AMPDU time spacing: 2 usec (0x04)
                HT TX/RX MCS rate indexes supported: 0-7, 32
                Bitrates (non-HT):
                        * 1.0 Mbps
                        * 2.0 Mbps (short preamble supported)
                        * 5.5 Mbps (short preamble supported)
                        * 11.0 Mbps (short preamble supported)
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
[... 64 lines cut ...]
        max scan plan interval: -1
        max scan plan iterations: 0
        Supported TX frame types:
                 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * mesh point: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
        Supported RX frame types:
                 * IBSS: 0x40 0xb0 0xc0 0xd0
                 * managed: 0x40 0xb0 0xd0
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * mesh point: 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        Supported extended features:
                * [ RRM ]: RRM
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211`

kismetwireless, 2021-06-14:
‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐
On Sunday, June 13th, 2021 at 9:06 PM, zekrifon ***@***.***> wrote:

>> You didn't provide much context, but: ERROR: Source mon1 (5FE308BD-0000-0000-0000-000F60048CE5) has encountered an error (mon1 could not bring up capture interface 'mon1', check 'dmesg' for possible errors while loading firmware: failed to set flags on interface 'mon1': Operation not permitted) Either your user lacks the permissions (and you didn't install suidroot), or you're trying to do monitor mode on a stock rpi internal wifi interface without nexmon patches.
>> […](#)
>
> Hi, thanks for the reply. I'm a real beginner here, so please let me know what could provide more context.
>
> But, I did install kismet with sudo make suidinstall.
>
> And I'm using an external WiFi adapter not the internal RPi WiFi

Still no real context about what you have or what drivers you're using; also whatever you're doing to paste output makes it extremely hard to read.

There are many wifi cards with many chipsets; some don't work. From the errors, I would guess whatever you've got - either the chipset, the drivers you picked or something, doesn't support monitor mode, but that's only a guess.

zekrifon, 2021-06-14:
I'm really sorry about the format. I messed up. 

I tried two wifi adapters:
[TP-Link Wireless N300 ](https://www.amazon.ca/TP-Link-TL-WN822N-Wireless-External-Antennas/dp/B00416Q5KI/ref=mp_s_a_1_14?dchild=1&keywords=tp+link+wifi+adapter&qid=1623636316&sprefix=tp+link+wifi+&sr=8-14)

And the [Canakit dongle](https://www.amazon.ca/CanaKit-Raspberry-Wireless-Adapter-Dongle/dp/B00GFAN498/ref=mp_s_a_1_3?dchild=1&keywords=raspberry+pi+wifi+dongle&qid=1623636227&sr=8-3)

I'm not sure I Installed any drivers in particular, though I followed these guides to compile, install and start kismet on boot:
[Kimset compiling quickstart](https://www.kismetwireless.net//docs/readme/quickstart/)
[How to Setup a Raspberry Pi Network Scanner](https://pimylifeup.com/raspberry-pi-network-scanner/)