kismetwireless/kismet #381: Current endpoint providing "kismet.device.packets_rrd"

notaco, 2021-06-30:
In 2020.12.R3 and prior "kismet.device.packets_rrd" is provided in /system/status.json. I have compiled 2021.06.R1 and 2021.05.R1 where it appears in tracked_fields.html but not in status.json. I saw a number of commits regarding changes to API so I was wondering if it has been relocated. 

Thanks!

kismetwireless, 2021-07-01:
Looks like that one slipped through the cracks on the docs; added.

https://www.kismetwireless.net/docs/devel/webui_rest/system_status/#packet-counts

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

notaco, 2021-07-01:
Thanks for clarifying. Noticed packetchain in commit titles and relevant RRDs coming through the websockets in the gui. Glad to see they are still available in the json endpoint.

closed 2021-07-01 by notaco