kismetwireless/kismet #385: disabling ncurses wrapper not working

freder, 2021-07-10:
the docs (https://www.kismetwireless.net/docs/readme/starting_kismet/#starting-kismet) and `kismet -h` say two different things (`--no-ncurses` vs. `--no-ncurses-wrapper`) – both don't seem to work.

- os: linux
- kismet version: `master` branch (`23f0f81ba7914f49f52fdd3b03cdf1c7f93f2d9e`)

kismetwireless, 2021-07-11:
both work and disable the ncurses - perhaps there's some confusion about what that means; you'll still see the text output from kismet; there is no way to disable that but if you really hate it you can redirect the output like w/ any other terminal program. Disabling ncurses will turn off the scroll manipulation and header bar telling you to go to the webui, and just print normal text output.

If you've already run something which modifies the terminal you may need to do 'stty sane' and 'reset' to re-initialize it.

-m

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

freder, 2021-07-11:
not sure what it was in the end, but after wiping everything and re-building from scratch it works now. -- thanks for your help!

closed 2021-07-11 by freder