kismetwireless/kismet #386: Web UI: assets (*.js, *.css) don't load

freder, 2021-07-10:
kismet version: master branch (23f0f81ba7914f49f52fdd3b03cdf1c7f93f2d9e)

on http://localhost:2501/ I get `404` errors for all included files, à la `GET http://localhost:2501/js/msgpack.js net::ERR_ABORTED 404 (Not Found)`

I had upgraded from an older version. removing `~/.kismet` and `/urs/local/etc` didn't change anything.

kismetwireless, 2021-07-10:
You've got an old config file or aren't calling the kismet binary you think you're calling; The kismet_httpd.conf file defines the path it uses for the http data.

-------- Original Message --------

freder, 2021-07-11:
specifying `--datadir /usr/local/share/` didn't work, but running kismet as root did. → it turned out to be a permissions problem on `/usr/local/share/kismet/httpd` (probably from the previous installation). _my bad!_ + thanks

closed 2021-07-11 by freder