kismetwireless/kismet #387: Q (not a bug).. RE: Naming convention of the kismet "group name".

microimage, 2021-07-11:
**Greeting valued team members. Thank you for taking my call.**
I really have no business asking this question as I am no where near a Git-Die quite yet. And Yoda ain't answering my calls. I just ran out of G-queries :O\

**RE: Is kismet capitalizing the 1st letter of it's group-name and Mint is taking offense**?

My "make install" is protesting a bit at the end. I know I can do 99 things wrong and only 1 right... Not sure if this [is] the 1 or the 99. Near the end the of "make install" process, are these last four (4) lines. The "**invalid group Kismet**" has me scratching my head.

sudo make install
_(time passes..)_
/usr/bin/install: **invalid group ‘Kismet’**
make[1]: *** [Makefile:470: commoninstall] Error 1
make[1]: Leaving directory '/home/vtm/bin/kismet'
make: *** [Makefile:612: install] Error 2
(end of make build)

Do i think it's possible that kismet prefers to capitalize the first letter of its (Kismet) Group name and my Mint is saying "fat chance kid"? Could be...Maybe.. If I try to "force the issue" with **addgroup --force-badname**. It gets a bit testee. And still throws up the error near the end of the build if I try to fake it out. So it has the appearances of Mint is yelling: "get off my case" when I try. See next.

$ sudo addgroup **K**ismet
addgroup: Please enter a username matching the regular expression configured
via the NAME_REGEX[_SYSTEM] configuration variable.  Use the `--force-badname'
option to relax this check or reconfigure NAME_REGEX.

_Any thoughts on the matter will be appreciated. I can live on bread crumbs, hints and innuendo. So spoon feeding is not my goal. Installing a prebuild is Plan B:. The one I had was "years" old.. Wanted to jump into the future!_

- **Mint:** Linux 5.4.0-77-generic 86-Ubuntu SMP Thu Jun 17 02:35:03 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

- **kismet:** Kismet 2021-00-GIT

- **Compiler**: gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

Thanks all. You RoCk (yeah, even without a answer:)

kismetwireless, 2021-07-11:
It shouldn't be capitalized - it's not in the install code; did you pass something odd to the configure stage? The group is always lowercase 'kismet' by default.

Upper cased letters in usernames and groups are generally considered bad; while some things might accept them you usually have to do special work to make that happen.

-m

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

microimage, 2021-07-12:
**Thank you for the reply. 
Clearly “Not all superheroes wear capes and fly”.**

RE: **Don't POP a CAP in it**
Super to hear that "it was not intended to be capitalized" as I was assuming. So it's possible I "assumed" this by a small set of random circumstances presented to me in a process I was unfamiliar with? No, newbies never do that! 

RE: **"Passing of anything"**.. except for a bastard kidney stone several years ago, nada. But that is InTeReSTiNg and that did not cross my mind at all :O\   5 points: kMan.

**Very COoL**!! This releases me from my self-imposed-brain-fart-imprisonment. I have different things to try now. I'll certainly take a closer look at "the basics", again. And start poking around the bush (usually reserved for 3rd date but any port in a storm)

_**Possibly interesting**; When I was posting this Q originally, I copied & pasted the error re: [**K**]ismet directly from the CLI.. The Capital [K] was present. But after pasting into Git it then displayed a lowercase [**k**]ismet. I chalked it up to a Git newbie not grasping the mechanics and subtitles of the process._ _(this will be a recurring theme)._

I'm "stoked" (can I still say that?) to get some new inspiration and look forward to finding out WHAT THE HE11 I DID. I'll post a follow up if there is  breaking news that might be helpful to the hive. 

Insightful "insights" and new facts to ponder. A Gr8T time to be alive. Clearly, from everything I can tell... 
"you are" The "_**kismet whisperer**_"

**Thank you for the Professional input and positive reinforcements.**

closed 2021-07-16 by microimage