kismetwireless/kismet #390: GPS not working

LuxStatera, 2021-07-27:
I have setup gpsd with /dev/ttyUSB0 and setup the kismet.config file with the gpsd setting for gps but it still will not connect. ive tried all sorts or settings but nothing. cgps shows my gps dongle is working. is anyone using a gps dongle and has a config file i can see? ive looked through all the documentation but there isnt much on gps and gpsd. also checked all sorts of forums but they are outdated.

Knuust, 2022-02-01:
Same here. I am using the Transsystem GPS/Glonass/Galileo/BeiDou that is recommended on https://www.kismetwireless.net/hardware/.
The device is shown as /dev/ttyUSB0 and I try to activate it with
`gps=serial:device=/dev/ttyUSB0,name=anyname,baud=9600`
but it won't work. The kismet UI says "unkown" and there is no gps data...

Can anyone help here?

SecretSantaa, 2022-05-02:
1. In /etc/kismet/kismet.conf, I added this under "GPS Configuration":
`gps=gpsd:host=localhost,port=2947,reconnect=true`

2. I start by GPS device by running:
`gpsd -n -N -D 2 /dev/ttyUSB0`

3. I verify I can get GPS info from:
`cgps -s`

4. Once I verify the GPS is up and working I start kismet with my wireless interface
`kismet -c wlan0`

Then in the top right corner of Kismet's web interface I can see the coordinates. Hope this helps!