kismetwireless/kismet #392: Not showing any devices on ARM64 build

curious-ninja, 2021-09-07:
Hey, hate to bug you again..

After the changes made to bring back the historical cloud (https://github.com/kismetwireless/kismet/issues/391), kismet doesn’t show any devices.

Works perfectly fine on my 64-bit VM Kali, devices show up and historical data is in the JSON. But doesn’t work on arm64 builds (Pi4 Kali): no devices ever show up, but it’s seeing packets because the upper right corner packet log is moving.

I am using nightly git builds.
I installed an older version (last release) and it worked fine on the Pi4.

Thanks

closed 2021-09-08 by kismetwireless

YO-BUDDY-YO, 2021-09-09:
Hey there!

I just pulled in the nightly build, enabled the history cloud logging, and attempted to run kismet. Unfortunately, the program crashes. I have tested on a Raspberry Pi 4 and a VM. Only happens on the Pi.

During the launch, it successfully sets the datasource (I've tried wlan0 and a previous log file), and immediately after, it errors out with FATAL:  Error during write(): Broken pipe.

Thanks for the support

kismetwireless, 2021-09-09:
I'll need more information than that, sorry - Which nightly, what's the full text of when it exits, what sort of devices are you using, what raspberry pi distribution?

If you can use the debug guide to provide a more complete dump of the crash, that will help: https://www.kismetwireless.net/docs/readme/debugging/

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

curious-ninja, 2021-09-16:
Works properly for me, using nightly packages.
History cloud is enabled and present in the JSON.

Keep in mind that when you compile, the configs are in /usr/local/etc/kismet*.conf instead of /etc/kismet/kismet*.conf (if using packages).

commit 642f0b26 (2021-09-08): Further optimizations for checksum, fix on some platforms.

Further optimizations for checksum, fix on some platforms.
Should fix #392 when it hits nightly builds