kismetwireless/kismet #394: Encountering a fatal error

FRobertAllen, 2021-10-05:
I'm not quite sure what to do?

INFO: Registered PHY handler 'UAV' as ID 4
INFO: Registered PHY handler 'NrfMousejack' as ID 5
INFO: Using default rates of 10/min, 1/sec for alert 'BLEEDINGTOOTH'
INFO: Registered PHY handler 'BTLE' as ID 6
INFO: Registered PHY handler 'RTLAMR' as ID 7
INFO: Indexing ADSB ICAO db
INFO: Completed indexing ADSB ICAO db, 322362 lines 6448 indexes
INFO: Registered PHY handler 'RTLADSB' as ID 8
INFO: Registered PHY handler '802.15.4' as ID 9
INFO: Registered PHY handler 'RADIATION' as ID 10
INFO: Could not open system plugin directory (/usr/lib/aarch64-linux-gnu/ki
      smet/), skipping: No such file or directory
INFO: Did not find a user plugin directory (/root/.kismet//plugins/),
      skipping: No such file or directory
INFO: GPS track will be logged to the Kismet logfile
ERROR: Error reading config file '/root/.kismet/kismet_httpd.conf': No
       such file or directory
LOCAL: This is the first time Kismet has been run as this user.  You will
       need to set an administrator username and password before you can
       use any features of Kismet.  Visit http://localhost:2501/ to
       configure the initial login, or consult the Kismet documentation at
       https://www.kismetwireless.net/docs/readme/webserver/ about how to
       set a password manually.
ERROR: (HTTPD) Could not read session data file, skipping loading saved
       sessions.
INFO: Serving static file content from /usr/share/kismet/httpd/
INFO: Enabling channel hopping by default on sources which support channel
      control.
INFO: Setting default channel hop rate to 5/sec
INFO: Enabling channel list splitting on sources which share the same list
      of channels
INFO: Enabling channel list shuffling to optimize overlaps
INFO: Sources will be re-opened if they encounter an error
INFO: Saving datasources to the Kismet database log every 30 seconds.
INFO: Launching remote capture server on 127.0.0.1 3501
/devices/views/all/last-time/:timestamp/devices

*** KISMET IS SHUTTING DOWN ***

*** KISMET HAS ENCOUNTERED A FATAL ERROR AND CANNOT CONTINUE.  ***
Shutting down plugins...
WARNING: Kismet changes the configuration of network devices.
         In most cases you will need to restart networking for
         your interface (varies per distribution/OS, but
         typically one of:
         sudo service networking restart
         sudo /etc/init.d/networking restart
         or
         nmcli device set [device] managed true

Kismet exiting.

I'm pretty new with at this by the way.

CyberGarrett, 2021-12-30:
I'm having the same issue, same output, running a fully updated kali VM and using an adapter that works with airmon-ng.

ChuckNorrison, 2022-07-28:
Same issue happened to me after cleanup all kismet files from system and reinstalled. Iam also missing the suid-install part after installation of kismet.