kismetwireless/kismet #397: RPi4 PAU09 N600 (RT5572) kismet_cap_linu Tainted

hexkey, 2021-11-05:
While kismet is running and hopping channels, following logs are filling syslog:

```
Nov  5 10:44:33 kali kernel: [  656.405491] ------------[ cut here ]------------
Nov  5 10:44:33 kali kernel: [  656.410165] WARNING: CPU: 2 PID: 920 at drivers/net/wireless/ralink/rt2x00/rt2x00config.c:188 rt2x00lib_config+0xfc/0x47c [rt2x00lib]
Nov  5 10:44:33 kali kernel: [  656.422173] Modules linked in: xfs r820t rt2800usb binfmt_misc rt2800lib rtl2832 rt2x00usb rt2x00lib i2c_mux regmap_i2c mac80211 dvb_usb_rtl28xxu sg cfg80211 dvb_usb_v2 dvb_core rfkill crc_ccitt libarc4 bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) raspberrypi_hwmon mc rpivid_mem uio_pdrv_genirq uio drm fuse drm_panel_orientation_quirks ip_tables x_tables ipv6
Nov  5 10:44:33 kali kernel: [  656.466937] CPU: 2 PID: 920 Comm: kismet_cap_linu Tainted: G        WC        5.4.83-Re4son-v8l+ #1
Nov  5 10:44:33 kali kernel: [  656.475975] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
Nov  5 10:44:33 kali kernel: [  656.481802] pstate: 60000005 (nZCv daif -PAN -UAO)
Nov  5 10:44:33 kali kernel: [  656.486590] pc : rt2x00lib_config+0xfc/0x47c [rt2x00lib]
Nov  5 10:44:33 kali kernel: [  656.491900] lr : rt2x00mac_config+0x54/0x90 [rt2x00lib]
Nov  5 10:44:33 kali kernel: [  656.497115] sp : ffffffc011a13650
Nov  5 10:44:33 kali kernel: [  656.500421] x29: ffffffc011a13650 x28: ffffff80df135940
Nov  5 10:44:33 kali kernel: [  656.505726] x27: 0000000000000014 x26: ffffff80f5972000
Nov  5 10:44:33 kali kernel: [  656.511031] x25: ffffffc010f86b80 x24: ffffff80eeb24000
Nov  5 10:44:33 kali kernel: [  656.516335] x23: ffffff80eeb263e8 x22: 0000000000000040
Nov  5 10:44:33 kali kernel: [  656.521639] x21: 0000000000000042 x20: ffffff80eeb24780
Nov  5 10:44:33 kali kernel: [  656.526943] x19: ffffff80eeb25e20 x18: 0000000000000000
Nov  5 10:44:33 kali kernel: [  656.532248] x17: 0000000000000000 x16: 0000000000000000
Nov  5 10:44:33 kali kernel: [  656.537551] x15: 0000000000000000 x14: 000000000000ba7e
Nov  5 10:44:33 kali kernel: [  656.542855] x13: 0000a06a1ac88e44 x12: 0000000000000000
Nov  5 10:44:33 kali kernel: [  656.548159] x11: 000000000000a06a x10: 00000000000019a0
Nov  5 10:44:33 kali kernel: [  656.553463] x9 : ffffffc011a132c0 x8 : ffffff80df137340
Nov  5 10:44:33 kali kernel: [  656.558767] x7 : 0000000000000014 x6 : 000000000000003e
Nov  5 10:44:33 kali kernel: [  656.564071] x5 : 00000000000000a3 x4 : ffffffc008dd2940
Nov  5 10:44:33 kali kernel: [  656.569375] x3 : ffffffc008dd2e04 x2 : 000000000000003d
Nov  5 10:44:33 kali kernel: [  656.574678] x1 : 000000000000003e x0 : 000000000000003e
Nov  5 10:44:33 kali kernel: [  656.579983] Call trace:
Nov  5 10:44:33 kali kernel: [  656.582426]  rt2x00lib_config+0xfc/0x47c [rt2x00lib]
Nov  5 10:44:33 kali kernel: [  656.587387]  rt2x00mac_config+0x54/0x90 [rt2x00lib]
Nov  5 10:44:33 kali kernel: [  656.592324]  ieee80211_hw_config+0x1d0/0x430 [mac80211]
Nov  5 10:44:33 kali kernel: [  656.597576]  ieee80211_set_monitor_channel+0x100/0x110 [mac80211]
Nov  5 10:44:33 kali kernel: [  656.603725]  cfg80211_set_monitor_channel+0x44/0x1cc [cfg80211]
Nov  5 10:44:33 kali kernel: [  656.609664]  __nl80211_set_channel+0xb0/0x2a0 [cfg80211]
Nov  5 10:44:33 kali kernel: [  656.614994]  nl80211_set_wiphy+0x264/0x910 [cfg80211]
Nov  5 10:44:33 kali kernel: [  656.620044]  genl_family_rcv_msg+0x1a4/0x3d4
Nov  5 10:44:33 kali kernel: [  656.624306]  genl_rcv_msg+0x64/0xdc
Nov  5 10:44:33 kali kernel: [  656.627786]  netlink_rcv_skb+0x60/0x12c
Nov  5 10:44:33 kali kernel: [  656.631613]  genl_rcv+0x40/0x5c
Nov  5 10:44:33 kali kernel: [  656.634746]  netlink_unicast+0x1c4/0x264
Nov  5 10:44:33 kali kernel: [  656.638660]  netlink_sendmsg+0x1a4/0x34c
Nov  5 10:44:33 kali kernel: [  656.642576]  sock_sendmsg+0x60/0x70
Nov  5 10:44:33 kali kernel: [  656.646057]  ____sys_sendmsg+0x21c/0x230
Nov  5 10:44:33 kali kernel: [  656.649971]  ___sys_sendmsg+0x90/0xe0
Nov  5 10:44:33 kali kernel: [  656.653625]  __sys_sendmsg+0x78/0xd0
Nov  5 10:44:33 kali kernel: [  656.657191]  __arm64_sys_sendmsg+0x30/0x40
Nov  5 10:44:33 kali kernel: [  656.661282]  el0_svc_common.constprop.0+0x7c/0x1b4
Nov  5 10:44:33 kali kernel: [  656.666065]  el0_svc_handler+0x38/0xac
Nov  5 10:44:33 kali kernel: [  656.669805]  el0_svc+0x8/0x204
Nov  5 10:44:33 kali kernel: [  656.672851] ---[ end trace 11bcb47641a970db ]---

# uname -a
Linux kali 5.4.83-Re4son-v8l+ #1 SMP PREEMPT Sat May 15 07:55:31 UTC 2021 aarch64 GNU/Linux

# lsusb
Bus 001 Device 004: ID 148f:5572 Ralink Technology, Corp. RT5572 Wireless Adapter

# kismet --version
Kismet 2021-08-R1

```

Any ideas?

kismetwireless, 2021-11-05:
yeah - basically the drivers for that chipset aren't great about changing channel.

not much you can do besides ignore the message; you could get a different type of card (mt7621 performs well, for instance) but there's probably no reason to be overly concerned.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

hexkey, 2021-11-08:
huh, alright thanks