kismetwireless/kismet #404: KISMET_URI_PREFIX undefined

Tibladar, 2022-01-01:
I set up a reverse proxy via this guide https://www.kismetwireless.net/docs/readme/webserver/#using-proxies-and-forwarding

`httpd_uri_prefix` is set but in javascript `KISMET_URI_PREFIX` is always undefined so websocket requests fail with 404 since the url-prefix is missing
[image]

[image]

OS: Ubuntu 20.04
Browser: Firefox 95.0.2
Kismet: Fresh 2021-08-R1 from official repo

kismetwireless, 2022-01-01:
That config value is only what the kismet *server* allows, it doesn't change the UI code.

Unfortunately because of how the websocket URIs have to be formed I doubt it's possible to proxy a nonstandard path anymore; I'll remove it from the docs as an option unless someone has a way to dynamically determine an unknown url path.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

kismetwireless, 2022-01-01:
The more I consider it, it may be possible with response header manipulation in nginx, I’ll add it to the list to investigate.

kismetwireless, 2022-01-02:
After much jackassery with javascript not being able to pull headers from the *current* request, the latest git commit plus the changes in the readme for how to set up a proxy (changing the proxy_set_header and adding a new add_header line), it should load the websocket via a proxy not homed in /.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

Tibladar, 2022-01-02:
Works fine now, thanks!

closed 2022-01-02 by Tibladar