kismetwireless/kismet #413: Webserver rewrite base / catchall fallback route

omnidepp, 2022-03-10:
Hi Everyone,
I'm building a kismet web plugin with vue as a single page application with routes.
For this to work properly, the kismet webserver would need to rewrite all virtual routes to /index.html. Is there an option to do this with the kismet internal webserver?

My SPA uses virtual routes that should be accessed like this: http://kismet_url:2501/plugin/plugin_name/device/12345.
So kismet webserver should catch all these virtual routes and rewrite them to ...plugin/plugin_name/index.html.

For webservers like apache and nginx, here is some documentation on how to accomplish this: https://router.vuejs.org/guide/essentials/history-mode.html#example-server-configurations

dragorn, 2022-10-11:
Currently there's nothing in Kismet to handle that; you could proxy it through an nginx that caught 404s and rewrite them in the interim; I'll put an optional catch-all config option on the list for future releases.

The kismet webserver isn't based on nginx/apache/etc or anything like that; it's a fully-integrated webserver that uses boost::beast as the base protocol handler, so there isn't any correlation between its configs and those of apache/nginx/etc.

-m

RFDeck, 2026-11-07:
The general version of this exists since 9180410b (2022-05-07): `httpd_redirect_unknown` in `kismet_httpd.conf`. When
no file matches a URL, Kismet serves the file given there instead of a 404. It's one setting for the whole server,
though. It doesn't give each plugin its own fallback under `/plugin/<name>/`, so an SPA plugin would need its own
index for that. If a per-plugin fallback is still wanted, that would be a feature request on top.