kismetwireless/kismet #414: Upgrade via RPI 3 (buster) - certificate issues

sfuthings, 2022-03-19:
I'm trying to upgrade Kismet 2016.07.R1-1 to a latest one on raspbian buster. Followed the instruction from https://www.kismetwireless.net/docs/readme/packages/ but says Kistmet is already the latest version. When I try to apt update i got this:

```
 sudo apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Ign:3 https://www.kismetwireless.net/repos/apt/release/buster buster InRelease
Err:4 https://www.kismetwireless.net/repos/apt/release/buster buster Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 165.227.176.149 443]
Reading package lists... Done
E: The repository 'https://www.kismetwireless.net/repos/apt/release/buster buster Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

kismetwireless, 2022-03-19:
This is unfortunately a debian / raspbian problem. About 4 months ago Lets Encrypt changed their signing certs, so distros need to update. For some reason despite a 6 month + notice from lets encrypt, debian and ubuntu both left this to the last minute it seems; others may have too, but the ubuntu one definitely hit me on my own systems too.

You need a new set of root ca certs - usually in the ca-certs package in raspbian/debian/ubuntu; assuming they've updated those appropriately for buster, you should be good to go.

I'll also mention that Buster is quite old as distributions go, and lacks a number of modern things Kismet needs, so some functionality is not available - as i recall, remote capture via websocket and all of the python3 services are disabled, because buster doesn't have usable versions of the support libraries. Ubuntu on pi, or the bullseye distro, would likely be what you want.

-m

sfuthings, 2022-03-26:
Agreed, definitely will update the rpi distro. Thanks.

closed 2022-03-26 by sfuthings