kismetwireless/kismet #415: PROBLEM LIB PYTHON3

vincenzogianfelice, 2022-03-21:
```
File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: KismetCaptureRtl433
Traceback (most recent call last):
  File "/usr/local/bin/kismet_cap_sdr_rtlamr", line 33, in <module>
    sys.exit(load_entry_point('KismetCaptureRtlamr==2020.10.1', 'console_scripts', 'kismet_cap_sdr_rtlamr')())
  File "/usr/local/bin/kismet_cap_sdr_rtlamr", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
Traceback (most recent call last):
  File "/usr/local/bin/kismet_cap_sdr_rtladsb", line 33, in <module>
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    sys.exit(load_entry_point('KismetCaptureRtladsb==2020.10.1', 'console_scripts', 'kismet_cap_sdr_rtladsb')())
  File "/usr/local/bin/kismet_cap_sdr_rtladsb", line 22, in importlib_load_entry_point
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: KismetCaptureRtlamr
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: KismetCaptureRtlads
```

these errors always appear when I try to go to the "Data Sources" item from the WebUI

kismetwireless, 2022-03-21:
You haven't properly install the datasources, you're missing the actual python modules.

Alternately, you changed your python base distribution and python interpreter, and did not re-install the datasources, so they're missing the python modules.

Either re-install the python components from kismet, or uninstall the python-based datasources (rtl433, adsb, amr, and any other python based plugins you installed).

If you change the system-wide python interpreter version, you need to re-install any dependencies under the new python version; lots of things will break otherwise.

vincenzogianfelice, 2022-03-21:
i tried installing via deb (via your repo) and it seems to work. One thing that still doesn't work is the opening of the openstreetmap.org site in the "ADSB Live" entry. The error comes out (connection denied). The site if I try to open it works correctly. TLS / SSL Problem?