kismetwireless/kismet #417: RTL_SDR Not Populating Data

Dinosaur-Ceiling, 2022-04-19:
Hello all, hoping I can find an answer to my issue. 

I have been trying to get my RTL_SDR (RTL2832U) to work in Kismet. I have installed the required RTL_433 and Kismet can see my RTL_SDR as an interface. I am able to enable the device, but no data is ever populated in Kismet. 

I have been able to run the SDR via just the rtl_433 code and have confirmed that it works to receive information. Just not sure how to get Kismet to see the data. 

Am I missing something simple that I just overlooked? 

Any help appreciated!

Marshall-Hallenbeck, 2022-04-24:
I just started playing around with the same RTL SDR (RTL2832U) and it worked fine for me, I just had to go into the "data sources" area in the web app and set it to "running"

[image]

Dinosaur-Ceiling, 2022-04-24:
Sorry, should have been more specfic. I am using the RTL-SDR driver application on Android which is a port of rtl_tcp to the platform which allows other applicatoions to start rtl_tcp and use the raw I/Q samples that it reads from USB DVB-T dongles based on the RTL2832U chip. The end goal being to use the RTL with an OTG cable to survey TPMS data and visualize it with the Kismet GUI. 

I have been able to get RTL_SDR data to populate in Kismet on a normal install on a Kali Linux computer. But I cannot get data to populate in Kali Nethunter. The Kali Nethunter Kismet can see the RTL_SDR USB driver, but from what I can tell, since I am running Kali Nethunter, it is not seeing any of the data being generated and would required a TCP passthrough. Not sure if that is possible or not.