kismetwireless/kismet #420: Problems using nightly builds repo on Ubuntu 20.04

peterlavelle, 2022-05-21:
Hi,

Was looking at trying out the nightly builds from the https://www.kismetwireless.net/repos/apt/git/focal apt repository but I'm having a problem using the signing key from https://www.kismetwireless.net/repos/kismet-release.gpg.key when trying to add the apt repo to my testing server, which uses Ubuntu 20.04

Error message I get when adding the repo is: Failed to update apt cache: E:The repository 'https://www.kismetwireless.net/repos/apt/git/focal focal Release' does not have a Release file.

Took a quick look at the gpg key locally (output below)

```
gpg --show-keys kismet-release.gpg.key 
pub   rsa4096 2018-09-12 [SC]
      ADA09A0E9B80ACCCE8FE6BB65345B8BF43403B93
uid                      Kismet Wireless (Packaging Signature) <kismet@kismetwireless.net>
sub   rsa4096 2018-09-12 [E]
```

Am I just being dumb and not using the correct signing key for the nightly builds repo?

kismetwireless, 2022-05-21:
It's possible you tried to pull just as it was updating? I just ran an apt update against focal nightly and it's fine currently...

Get:9 https://www.kismetwireless.net/repos/apt/git/focal focal/main amd64 Packages [4,521 B]
Get:10 https://www.kismetwireless.net/repos/apt/git/focal focal/main i386 Packages [1,874 B]
Fetched 136 kB in 1s (142 kB/s)
Reading package lists... Done

peterlavelle, 2022-05-21:
HI,

Tried just now and getting the same error

Failed to update apt cache: E:The repository 'https://www.kismetwireless.net/repos/apt/git/focal focal Release' does not have a Release file

peterlavelle, 2022-05-21:
Update: I've managed to save myself from my own idiocy. I deleted the repo key from apt and re-added it just now and that seemed to get things working again.

closed 2022-05-21 by peterlavelle

kismetwireless, 2022-10-11:
Ah - you have the wrong definition for the packages.

It's:

deb https://www.kismetwireless.net/repos/apt/git/focal focal main