kismetwireless/kismet #424: Adding and Env Var for SUID

g-h-97, 2022-08-05:
Hi,

It seems that kismet prompts the user for whether to activate setuid bit or not, and this causes an issues while installing kismet as part of Dockerfile RUN, where it causes the entire install to hang at that prompt.

Could you please consider adding an environment variable for it ? along the lines of SUIDBIT or somthing similar.

Thanx,

dragorn, 2022-08-05:
This is controlled by dpkg; you probably should be doing non-interactive installs in your dockerfiles;

DEBIAN_FRONTEND=noninteractive apt ...

g-h-97, 2022-08-05:
Thanx for the heads-up @dragorn, I'll try it as soon as I can. I don't really use debian or it's derivatives that much, since I use Arch BTW X)

I really appreciate you help, thank you,

closed 2022-08-05 by g-h-97