kismetwireless/kismet #426: GPS In GUI Remains Green/Lock and COORDS Do Not Refresh When GPS adapter disconnected

busysignal, 2022-08-08:
Nightly installed 2022-08-08, Seaberry Arm and Intel;

Configure Kismet for GPSD 
```
# GPS Settings
gps=gpsd:host=localhost,port=2947,reconnect=true
```
Leave GPS connected, gets lock, Kismet reports active and locked 
Disconnect GPS, Kismet GUI still reports green, still leaves last coords on the GUI.

k-rku, 2024-04-04:
This seems to be an older 'issue' so you probably don't need this anymore.
For anyone else reading this... update your gpsd version.

This specific issue seemed to be caused by gpsd version 3.22.

Here are the steps I took to update to gpsd version 3.25, which solved it for me:

**Stop possibly interfering gpsd services:**
sudo systemctl stop gpsd.socket
sudo systemctl disable gpsd.socket

**Here's the subsequent install:**

sudo apt install -y scons \ libncurses-dev python-dev-is-python3 pps-tools git-core asciidoctor python3-matplotlib\ build-essential manpages-dev pkg-config python3-distutils

wget http://download.savannah.gnu.org/releases/gpsd/gpsd-3.25.tar.gz
tar -xzf gpsd-3.25.tar.gz

cd gpsd-3.25/

sudo scons
sudo scons udev-install
sudo reboot

busysignal, 2024-04-05:
I never came back to this and will close issue. 
it is not 100% repro at this time and this is rather old.

closed 2024-04-05 by busysignal