kismetwireless/kismet #428: Request: please add bookworm to the repos and support for libwebsockets17

xevilstar, 2022-08-24:
apt-get install kismet
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 kismet : Depends: kismet-capture-linux-wifi but it is not installable
          Depends: kismet-capture-linux-wifi but it is not installable
          Depends: kismet-capture-linux-bluetooth but it is not installable
          Depends: kismet-capture-nrf-mousejack but it is not installable
          Depends: kismet-capture-ti-cc-2531 but it is not installable
          Depends: kismet-capture-ubertooth-one but it is not installable
          Depends: kismet-capture-nrf-51822 but it is not installable
          Depends: kismet-capture-nxp-kw41z but it is not installable
          Depends: kismet-capture-nrf-52840 but it is not installable
          Depends: kismet-capture-rz-killerbee but it is not installable
 kismet-capture-hak5-wifi-coconut : Depends: libwebsockets16 but it is not installable

libwebsockets-dev - lightweight C websockets library - development files
libwebsockets-evlib-ev - lightweight C websockets library
libwebsockets-evlib-glib - lightweight C websockets library
libwebsockets-evlib-uv - lightweight C websockets library
libwebsockets-test-server - lightweight C websockets library - test servers
libwebsockets-test-server-common - lightweight C websockets library - test servers data
libwebsockets17 - lightweight C websockets library

xevilstar, 2022-08-24:
> bookworm is the next debian version (the one after bullseye) and it comes
> with libsockets17, but anyway thanks for the kali tip ;)

upps....

The following packages have unmet dependencies:
 kismet-capture-linux-bluetooth : Depends: libwebsockets16 but it is not
installable
 kismet-capture-linux-wifi : Depends: libwebsockets16 but it is not
installable
 kismet-capture-nrf-51822 : Depends: libwebsockets16 but it is not
installable
 kismet-capture-nrf-52840 : Depends: libwebsockets16 but it is not
installable
 kismet-capture-nrf-mousejack : Depends: libwebsockets16 but it is not
installable
 kismet-capture-nxp-kw41z : Depends: libwebsockets16 but it is not
installable
 kismet-capture-rz-killerbee : Depends: libwebsockets16 but it is not
installable
 kismet-capture-ti-cc-2531 : Depends: libwebsockets16 but it is not
installable
 kismet-capture-ti-cc-2540 : Depends: libwebsockets16 but it is not
installable
 kismet-capture-ubertooth-one : Depends: libwebsockets16 but it is not
installable

***@***.***:/etc/apt/sources.list.d# cat kismet.list
deb [arch=arm64]  https://www.kismetwireless.net/repos/apt/release/kali
kali main

kismetwireless, 2022-08-24:
Ah - so you're trying to install the bullseye packages on a newer debian. I'll see if I can pull docker build images for it.

xevilstar, 2022-08-24:
Awesome, thanks!

xevilstar, 2022-08-24:
https://github.com/cisagov/docker-debian12-ansible'

or install a bullseye and put this in the apt sources.list

deb http://deb.debian.org/debian bookworm main contrib non-free
deb-src http://deb.debian.org/debian bookworm main contrib non-free #Added by software-properties
deb http://security.debian.org/debian-security bookworm-security main contrib non-free
deb-src http://security.debian.org/debian-security bookworm-security main contrib non-free #Added by software-properties
deb http://deb.debian.org/debian bookworm-updates main contrib non-free
deb-src http://deb.debian.org/debian bookworm-updates main contrib non-free #Added by software-properties

then the usual apt-get update && apt-get full-upgrade
then a reboot

dragorn, 2022-08-24:
------- Original Message -------
On Wednesday, August 24th, 2022 at 11:52 AM, xevilstar ***@***.***> wrote:

> https://github.com/cisagov/docker-debian12-ansible'

I've got a pretty complex build system already which relies on docker and virtualized containers for everything, because I'm building packages for a dozen different distros; I don't have the systems (or storage) to run dedicated hardware for all the different distros.

That said it looks like integrating bookworm into the list isn't too much of a hassle; if you want to try the first build of the nightly packages they're at:

'deb https://www.kismetwireless.net/repos/apt/git/bookworm bookworm main'
Looks like bookworm mysteriously dropped libubertooth, so those aren't part of the build.

>

xevilstar, 2022-08-25:
> thanks a bunch !!!!!!!!
>
> Now I will try to install
>
> just one note ... on raspberrypi4 I had to add
>
> deb [arch=arm64] https://www.kismetwireless.net/repos/apt/git/bookworm bookworm
> main
>
>
> to avoid the error:
> Fetched 9,393 kB in 4s (2,144 kB/s)
> Reading package lists... Done
> N: Skipping acquire of configured file 'main/binary-armhf/Packages' as
> repository 'https://www.kismetwireless.net/repos/apt/git/bookworm
> bookworm InRelease' doesn't support architecture 'armhf'

I have libubertooth in my apt-list

apt-cache search libubertooth
libubertooth-dev - Shared library for Bluetooth experimentation -
development files
libubertooth1 - Shared library for Bluetooth experimentation

cat sources.list
deb http://deb.debian.org/debian bookworm main contrib non-free
deb-src http://deb.debian.org/debian bookworm main contrib non-free #Added
by software-properties
deb http://security.debian.org/debian-security bookworm-security main
contrib non-free
deb-src http://security.debian.org/debian-security bookworm-security main
contrib non-free #Added by software-properties
deb http://deb.debian.org/debian bookworm-updates main contrib non-free
deb-src http://deb.debian.org/debian bookworm-updates main contrib non-free
#Added by software-properties
# Uncomment deb-src lines below then 'apt-get update' to enable 'apt-get
source'
# deb-src http://deb.debian.org/debian bullseye main contrib non-free
# deb-src http://security.debian.org/debian-security bullseye-security main
contrib non-free
# deb-src http://deb.debian.org/debian bullseye-updates main contrib
non-free

xevilstar, 2022-08-25:
I can confirm that those packages install a working release.
I am a kismet n00b but in the web interface I can see some bluetooth output too ...
now I have just to wait for the nexmon devs to support more recent kernels like 5.18 5.19 or (in my case)
 uname -a
Linux raspberrypi 6.0.0-rc1-v8+ #2 SMP PREEMPT Fri Aug 19 14:48:09 CEST 2022 aarch64 GNU/Linux
I could try to install their firmware as it is but I am frightened to brick my poor pi

xevilstar, 2022-10-11:
> by the way .... if you install a bullseye and edit the apt sources-list
> run apt-get update && apt-get full-upgrade
> what you get is a bookworm (I think that's feasible with docker too)

by the way ....
look here https://github.com/cisagov/docker-debian12-ansible

xevilstar, 2022-10-11:
bookworm is the next debian version (the one after bullseye) and it comes
with libsockets17, but anyway thanks for the kali tip ;)

xevilstar, 2022-10-11:
by the way .... if you install a bullseye and edit the apt sources-list

deb http://deb.debian.org/debian bookworm main contrib non-free
deb-src http://deb.debian.org/debian bookworm main contrib non-free #Added
by software-properties
deb http://security.debian.org/debian-security bookworm-security main
contrib non-free
deb-src http://security.debian.org/debian-security bookworm-security main
contrib non-free #Added by software-properties
deb http://deb.debian.org/debian bookworm-updates main contrib non-free
deb-src http://deb.debian.org/debian bookworm-updates main contrib non-free
#Added by software-properties

run apt-get update && apt-get full-upgrade

what you get is a bookworm (I think that's feasible with docker too)

joecupano, 2023-09-01:
> Ah - so you're trying to install the bullseye packages on a newer debian. I'll see if I can pull docker build images for it.
> […](#)

For what it's worth I ran into the "libwebsockets17 not available" in the following:

- Raspberry Pi OS (64-bit) current release on Raspberry Pi 4B 4GB RAM
- Ubuntu 22.04 LTS (and64) 4GB RAM running as QEMU/KVM

I am sourcing from https://www.kismetwireless.net/repos/apt/release/kinetic kinetic main  repo

pdobrien3, 2023-11-11:
so, I read the entire thread.  I can't figure out if anyone has successfully setup the capture tools on a bookwork raspi install?  If so, how?

SFLx3c4, 2024-05-07:
I'm running on a Raspberry Pi 4 Model B Rev 1.5
By adding arch=arm64 the following it worked for me:

https://www.kismetwireless.net/packages/#debian-bookworm

wget -O - https://www.kismetwireless.net/repos/kismet-release.gpg.key --quiet | gpg --dearmor | sudo tee /usr/share/keyrings/kismet-archive-keyring.gpg >/dev/null
echo 'deb [***arch=arm64*** signed-by=/usr/share/keyrings/kismet-archive-keyring.gpg] https://www.kismetwireless.net/repos/apt/release/bookworm bookworm main' | sudo tee /etc/apt/sources.list.d/kismet.list >/dev/null
sudo apt update
sudo apt install kismet

Thanks for the inputs from xevilstar

thebaldgeek, 2024-11-07:
Continuing the bookworm install issue.
Fresh Pi4 headless install of Bookworm and following the instructions in the post above results in the following:   

` $ sudo apt install kismet
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 kismet:arm64 : Depends: kismet-core:arm64 but it is not going to be installed
                Depends: kismet-capture-linux-wifi:arm64 but it is not going to be installed
                Depends: kismet-capture-linux-wifi:arm64 but it is not going to be installed
                Depends: kismet-capture-linux-bluetooth:arm64 but it is not going to be installed
                Depends: kismet-capture-nrf-mousejack:arm64 but it is not going to be installed
                Depends: python3-kismetcapturertl433:arm64 but it is not installable
                Depends: python3-kismetcapturertladsb:arm64 but it is not installable
                Depends: python3-kismetcapturertlamr:arm64 but it is not installable
                Depends: python3-kismetcapturefreaklabszigbee:arm64 but it is not installable
                Depends: kismet-capture-ti-cc-2531:arm64 but it is not going to be installed
                Depends: kismet-capture-nrf-51822:arm64 but it is not going to be installed
                Depends: kismet-capture-nxp-kw41z:arm64 but it is not going to be installed
                Depends: kismet-capture-nrf-52840:arm64 but it is not going to be installed
                Depends: kismet-capture-rz-killerbee:arm64 but it is not going to be installed
                Depends: kismet-capture-hak5-wifi-coconut:arm64 but it is not going to be installed
                Depends: kismet-logtools:arm64 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
`

The only way I have found (after about 40 hours working on this over the past 2 weeks) is to build kismet from source.
However, if you do that, GPS will not work... So I'm still dead in the water, but have learned a ton.   
   
Has anyone a reliable non-docker way to install kismet on a headless Pi4 bookworm?