kismetwireless/kismet #432: WebUI - No device highlighting and alert indicator not working

RompitaKablo, 2022-09-11:
I previously used an older version of Kismet with the web interface. I installed this by adding the Kismet repo to my sources list.

Recently this upgraded my copy of Kismet to, I believe, Kismet 2022-08-R1.

Since then if Kismet captures any handshakes the device line colour does not change even though Kismet is set to do so. Also if I flood my AP with deauth packets I can see the alerts in the terminal but no alerts appear in the web ui alert list or the log below the device list.

I get the same if I install a "fresh" copy of Kismet on another machine.

The device highlighting and alert indicator both worked on the previous version (sorry, I do not know what version that was).

[image]

Is this a bug or have I missed a new configuration step?

solsticedhiver, 2022-09-28:
I just tested a deauth flood with aireplay-ng, and the alerts are showing up in the messages window of the kismet web ui, here. Also in the alerts tab panel, and the icon changed to red too.

You should check first that you have a kismet_alert.conf file and that it is included in the main kismet.conf

What sure is strange is that you got the alerts in the ncurses console but not the web ui. Like you said, it could be just a configuration error. May be a missing file or something?

Does kismet say anything unusual when starting?

RompitaKablo, 2022-09-29:
Thanks for the reply.

I've been installing this on RasPiOS Bullseye x64 using the Kismet repo and each time I have had the same problem, even on a fresh copy of RasPiOS. I even downloaded Kismet from GitHub and compiled it, still the same. Yes there is a `kismet_alert.conf` file. No messages of note when starting Kismet. I have reverted to an older copy of Kismet that seems to work. Thanks for the info about **discord**.

closed 2023-03-01 by RompitaKablo