kismetwireless/kismet #436: Wi-fi client behavior running list

jonathanseeh, 2022-09-29:
Hi! Some strange things have been popping up when running kismet to monitor my home network. Recently I've been coming across a long long list of what looks like spoofed wifi client behavior. I can't tell if it's an outside attack on my AP, or if it's a configuration issue or bug. Can anyone help shed some light on what might be happening?  In a nutshell, it keeps picking up new APs with 00:00:00:00 every 20 seconds. 
[image]

solsticedhiver, 2022-10-12:
Could you be more specific?
What is the behavior you are seeing that seems to you an issue?
What are you expecting instead?

jonathanseeh, 2022-10-14:
My apologies.

I constantly get a new ap discovered alert, about 4-5 every minute, with what seems like the first four mac octets are 00:00:00:00 followed by a random last two octets, always changing, all different channels. It looks like this:

new ap discovered 00:00:00:00:C7:38
new ap discovered 00:00:00:00:E8:F8
and so on....thousands over the course of the day. it makes my little pi crash when the data gets too much.
I use a realtek rtl88xxau driver for my dongle.

I strongly think this may be related to my neighbors gaming setup, a mesh network wpa3 running 160mhz channels....Im capturing with my little ac dongle lol. is this the issue?

better yet is there a way i can filter out or ignore 00:00:00:00:*:* ?

ty in advance

Sent with [Proton Mail](https://proton.me/) secure email.

solsticedhiver, 2022-10-14:
I can't say if it's the *rtl88xxau* driver or anything else.

Yes, you can filter those out. like explained in the doc: https://kismetwireless.net/docs/readme/config_files/

You could try to use something like:

    kis_log_device_filter=IEEE802.11,any,00:00:00:00:00:00/ff:ff:ff:ff:ff:00:00,block
    kis_log_packet_filter=IEEE802.11,any,00:00:00:00:00:00/ff:ff:ff:ff:ff:00:00,block

This uses a matched group. I have never used that; not sure it will work for you. You have to check and try for yourself. This is explained in kismet_filter.conf. You can paste these changes in kismet_site.conf

You can close the issue, if it answers your question. You can ask other questions in the discord kismet server, room #support. Details at https://kismetwireless.net/#irc-discord